raspibolt/raspibolt #361: SD card filling up with data

Hostiletriangle, 2019-01-29:
My SD card is filling up with data instead of the hdd. How do I solve this?
[image]

Thanks in advance

Stadicus, 2019-01-29:
First thing would be to check what is filling up your disk space: blockchain data, logs, other stuff? For that, as "admin", please install `ncdu` and check your sdcard:

```
$ sudo apt install ncdu
$ cd /
$ ncdu
```
This is what I get:
```
ncdu 1.12 ~ Use the arrow keys to navigate, press ? for help 
-----------------------------------------------------------------------------
. 282.9 GiB [##########] /mnt
  721.5 MiB [          ] /usr
. 539.2 MiB [          ] /var
  351.1 MiB [          ] /lib
   84.7 MiB [          ] /boot
.  56.4 MiB [          ] /home
   13.6 MiB [          ] /media
.   5.9 MiB [          ] /run
.   5.7 MiB [          ] /etc
    5.6 MiB [          ] /bin
    5.4 MiB [          ] /sbin
.  44.0 KiB [          ] /tmp
!  16.0 KiB [          ] /lost+found
e   4.0 KiB [          ] /srv
!   4.0 KiB [          ] /root
e   4.0 KiB [          ] /opt
.   0.0   B [          ] /sys
.   0.0   B [          ] /proc
    0.0   B [          ] /dev
```
If you see a very big directory, use the arrow keys and [Enter] to investigate further.

Hostiletriangle, 2019-01-29:
My /var folder is much bigger than expected
[image]

turns out its the /log and swap
[image]

and especially the deamon and sys logs
[image]

But the swapfile is not supposed to be on my sd card right?

Stadicus, 2019-01-31:
Check active swap file (if it's not active, you can delete it):
```
$ cat /proc/swaps
Filename                                Type            Size    Used    Priority
/mnt/hdd/swapfile                       file            102396  98468   -2
``` 
The `daemon.log` is way too big. Try analyzing it to see what process produces this many entries (google might help) and paste some relevant findings here (prefereably using pastebin.com).

Hostiletriangle, 2019-02-01:
My swapfile was on the SD card instead of on the hdd. So I disabled the swapfile, put it on the hdd and deleted the old one. This also solved the problem with the big deamon.log for some reason.

closed 2019-02-01 by Hostiletriangle