raspibolt/raspibolt #362: Connecting lnd to an external btcpayserver

talentpierre, 2019-01-29:
hi.
i tried to connect my raspberry with lnd, set up with your tutorial, to an external btcpayserver
but it didn't work. i have a dynamic ip and tried solve this through duckdns. but i had to use a https:// format for btcpayserver. how can i get through it?
it would be great if you could cover it on the bonus section of your tutorial.

talentpierre, 2019-01-31:
1. yes, i  set up a RaspiBolt full node (bitcoind / lnd) at home
2. i use duckdns.org to handle the dynamic ip
3. yes, i am trying to
this is my btcpayserver on a vps through lunanode service
i have an connection URI but it is recommended to use the ways in the picture
[image]

Stadicus, 2019-01-31:
I'm not familiar with BTCPayServer, although it's definitely on my bucket list. Not sure how I can help here, that seems like a BTCPayServer configuration issue, don't you think? Or are you struggling with the full node end of the pairing? Sorry, it's still unclear to me what you want to achieve and how that should work.

talentpierre, 2019-01-31:
sorry for writing so unspecific. it's because i don't know what the actual problem is. i just want to connect my raspibolt to my btcpayserver.

if you look in the picture above you will see that i have to enter a connection string.
this string contains of
--> the type of connection (lnd-rest) --> easy
--> server (my https:// address to my node) --> some kind of difficult
--> my macaroons in hex format (easy)
--> my certthumbprint in hex format (easy)

i have to use a https connection to my node because the macaroons will be sent through it. if i don't use https my macaroons could be seen by a third party.
so i think the question is: how can i connect my raspibolt to my server using https?
my server has an https address so, how do i get one on the raspibolt?
do i have to install something like "letsencrypt"? do i have to change something in my .conf file?

this is the documentation for raspiblitz on this topic.
https://docs.google.com/document/d/1PyX4R6zQt8tyVl8wkWhphAfFU5q4HatBGv3YvvZX6zE/preview

thank you for your help

Stadicus, 2019-02-08:
Ah, so you basically just want to use the BTCPayServer web interface on your VPS, without the Bitcoin / LND part, outsourcing that to your Raspberry Pi?

Reading through the documentation of RaspiBlitz, you need to expose both ports 8332 (bItcoind) and both rpclisten and restlisten for lnd.

I would advice heavily advice against such a setup:
1. A Raspberry Pi is not made as an API backend. It's a critical single point-of-failure and not suitable for business usage.
2. Exposing port 8332 is against every best practice I know. This port is highly sensitive and allows everyone to remotely control your Bitcoin Core setup.
3. Not sure if that works with the self-signed HTTPS certificates of lnd. If not, you need to have a real domain name and use Let's Encrypt to generate a real TLS certificate. 

All in all, it seems more of a hassle and is bound to break than running everything on a VPS (as BTCPayServer is meant to be) for ~5$/month. Anyway, this configuration is definitely outside the scope of this "beginner's guide".

talentpierre, 2019-02-08:
ok. thanks a lot for your lengthy and helpful response. 
i really appreciate your help and didn't think about those consequences.

closed 2019-02-12 by Stadicus