raspibolt/raspibolt #366: Tor routing node

HelgeHunding, 2019-01-31:
First, thanks for a great guide!

Everything went smoothly until the Tor setup.

Before enabling Tor, everything worked. My full node was visible at https://bitnodes.earn.com/ and lightning was working. I could open channels from my desktop and my mobile phone (connected to other nodes).

Then I enabled Tor. My node was no longer visible at bitnodes (not when trying with https://bitnodes.earn.com/nodes/"address".onion-8333/ either). But my node was getting more then 8 connections and a few of them was inbound so I figured the issue was with Bitnodes (I've read the issues other people was having with this).

When I went on to lightning I got confused. I'm not sure if the Tor-guide is suppose to be for a routing node? Anyway, my node went live and was advertised as a public node at https://1ml.com/node/028b8ceb1e93526f6cf972640dd9ca0304e0d05f7f5cd437039c8179d4500d3d97

I could open channels and Bitrefill was, from what I could tell, able to open a channel with me. But when I tried to connect from my desktop or my mobile phone I kept getting an error like this:
`error connecting to peer: dial tcp: lookup v3lh355tcfdf5iabhdrwegq6vfmtq3g47unze2ujgj7udno2mc4bn5ad.onion: no such host
`
I can find my node in an application like Zap desktop wallet or on my phone, but I can't connect. I can connect to other nodes (both normal and .onion).

I tried looking at this page https://github.com/lightningnetwork/lnd/blob/4dd4f7cf452709d90fa1d16062b13874623c929a/docs/configuring_tor.md but I couldn't quite understand everything. Might be an issue with --tor.dns or --tor.socks?

I've tried a few settings, right now everything should be back to normal. I've disabled the UFW to rule it out.
Bitcoin conf:
`
# Bitcoind options
server=1
daemon=1

# Connection settings
rpcuser=xxxx
rpcpassword=xxxx
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

# Tor
proxy=127.0.0.1:9050
bind=127.0.0.1
listenonion=1
onlynet=onion
`

Lightning conf:
`[Application Options]
debuglevel=info
maxpendingchannels=5
alias=ABCLND
color=#68F442
listen=localhost

# Your router must support and enable UPnP, otherwise delete this line
nat=false

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

[Tor]
tor.active=true
tor.v3=true
`
tor/torrc:
`ControlPort 9051
CookieAuthentication 1
CookieAuthFileGroupReadable 1`

tor/tor-service-defaults-torrc:
`DataDirectory /var/lib/tor
PidFile /var/run/tor/tor.pid
RunAsDaemon 1
User debian-tor

ControlSocket /var/run/tor/control GroupWritable RelaxDirModeCheck
ControlSocketsGroupWritable 1
SocksPort unix:/var/run/tor/socks WorldWritable
SocksPort 9050

CookieAuthentication 1
CookieAuthFileGroupReadable 1
CookieAuthFile /var/run/tor/control.authcookie

Log notice syslog`

Any clarification or suggestion on what might be the issue? Have I misunderstood anything with running a routing node over tor?

Thanks!

Stadicus, 2019-01-31:
From a first glance, your configuration looks correct. I think this is behaviour as intended. If your node publishes its onion address, it's a routing node (otherwise just the pubkey would be visible). 

To initiate a connection to your node, other nodes must have Tor installed as well. This is likely not the case with your regular desktop or mobile wallet. I just tried from my end and I could connect and open a channel with you on the first try, even from within Zap Desktop.

[image]

[image]

If someone wants to open a channel, but can't due to Tor, you can connect from your side and the other person can the open the channel with the pubkey only (without the @.......onion).

HelgeHunding, 2019-02-01:
Thanks for the reply!

I tried to connect to my Pi-node when running tor on my desktop and indeed it worked. I'm still a bit confused on why I was able to connect to other .onion nodes without running tor on my desktop before. I have some reading to do there. 

Everything seems to work now, so closing this issue.

closed 2019-02-01 by HelgeHunding