raspibolt/raspibolt #368: Not Visible from Internet

larry-beck, 2019-02-04:
I have setup per guide here and all appears to be working fine, synced with blockchain, setup to allow in and out on all paths (ipv4, ipv6, onion). I connect to ipv4, ipv6, onion nodes just fine and they shows as peers in the list, but I am only able to receive inbound connections from ipv4 only, none connect via ipv6 or onion. When I try to test addresses for ipv6 and onion shown in my getnetworkinfo via bitnodes, they both show unreachable, ipv4 works fine and is reachable from bitnodes. Any ideas, what do you need to see to verify as I don't want to post too many configs if not needed, etc.

# RaspiBolt: bitcoind configuration
# /home/bitcoin/.bitcoin/bitcoin.conf

# remove the following line to enable Bitcoin mainnet
#testnet=1

# Bitcoind options
server=1
daemon=1

# Connection settings
rpcuser=xxx
rpcpassword=xxx

#onlynet=ipv4
#onlynet=onion
#onlynet=ipv6
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

# add / change as needed:
onion=127.0.0.1:9050
#proxy=127.0.0.1:9050
#bind=127.0.0.1
listenonion=1
listen=1
discover=1
externalip=XXXXXX.onion

## Configuration file for a typical Tor user
# /etc/tor/torrc

## The port on which Tor will listen for local connections from Tor
## controller applications, as documented in control-spec.txt.
ControlPort 9051
CookieAuthentication 1
CookieAuthFileGroupReadable 1

############### This section is just for location-hidden services ###
HiddenServiceDir /var/lib/tor/bitcoin-service/
HiddenServicePort 8333 127.0.0.1:8333

larry-beck, 2019-02-04:
I resolved the inbound ipv6 issue, now reachable from bitnodes site, but I cannot figure out a way to allow onion traffic through my router/firewall, AT&T Fiber, model 5268AC, since I cannot add a firewall rule for onion address like ipv4 or ipv6 which the router supports. Any ideas or workarounds?

larry-beck, 2019-02-07:
Still cannot get inbound onion traffic per bitnodes showing that I am reachable, hidden tor traffic from within Bitcoin Core is working fine.

Stadicus, 2019-02-08:
Bitnodes seems finicky with onion nodes. Did you try this approach?
https://github.com/Stadicus/guides/issues/189#issuecomment-449638447

janglorumo, 2019-02-13:
Hello! Loved the guide and have my pi running fine for the most part. Payments working, inbound and outbound channels working so far. My full node is still not visible even trying what was mentoined from #189 
gfbi4yc2e4r5ppfk.onion-8333/

Cause for concern?

Stadicus, 2019-02-21:
I don't know of a good way for network discovery over Tor, bitnodes seems very unreliable. As long as others can open channels to your node, it's definitely reachable and no worry for concern.

closed 2019-03-03 by Stadicus