raspibolt/raspibolt #369: Failed to stop lnd.service

jbrodskytd, 2019-02-04:
Hi I'm running into an issue when trying to use systemctl to stop my lnd.service

```
$ sudo systemctl stop lnd
Failed to stop lnd.service: Failed to activate service 'org.freedesktop.systemd1': timed out
See system logs and 'systemctl status lnd.service' for details.
Failed to get load state of lnd.service: Connection timed out
```

Any ideas? my node has been up and running for a few weeks now.

Stadicus, 2019-02-08:
Can you run through the "Troubleshoot" section and paste relevant log files here?

daniel-vera-g, 2019-02-10:
> Hi I'm running into an issue when trying to use systemctl to stop my lnd.service
>
> ```
> $ sudo systemctl stop lnd
> Failed to stop lnd.service: Failed to activate service 'org.freedesktop.systemd1': [...]

Was lnd.service running at all?

What does `systemctl status lnd.service` output?
What are the logs saying?

jbrodskytd, 2019-02-11:
Thanks guys, I ended up doing a hard restart of my raspibolt... none of the system commands would work but Lnd was running and I can run lncli commands. Seems to be working ok now. If I run into it again I'll do as Stadicus suggested. Thanks again!

closed 2019-02-12 by Stadicus