raspibolt/raspibolt #370: System Overview

Fillippone, 2019-02-05:
Hello,
a simple enanchment for your System Overview. 
I think it would be useful to display the actual number of incoming/outgoing conncections on our Bitcoin Node in the System Overview. It should be trivial to implement, but I am not able to do it by myself!

Stadicus, 2019-02-08:
Thanks for the suggestion, maybe someone will pick it up? @robclark56

Fillippone, 2019-02-16:
Anyone? I made some progress, but my programming skills aren’t definitely capable of this.

robclark56, 2019-02-17:
What bitcoin-cli command would display what you are after?

Fillippone, 2019-02-17:
It is a combination of Bitcoin-cli get network stat (to get all active connections) and bitcoin-cli getpeerinfo   select(.inbound == true)'  to get inbound connections.
My idea is getting a line reading Connections (I/O): 39 (10/29) to be displayed somewhere.

robclark56, 2019-02-17:
this should work
bitcoin-cli getpeerinfo | grep -c 'inbound": false'
bitcoin-cli getpeerinfo | grep -c 'inbound": true'

Will see what I can do

Fillippone, 2019-02-17:
Someone started working on that. He pulled the info’s right, but garbled the layout (he cannot test the output as he doesn’t have a raspberry to test:
https://github.com/inaltoasinistra/guides/blob/master/raspibolt/resources/20-raspibolt-welcome

robclark56, 2019-02-18:
Tell me if this works. If so ... I will make a pull request

```
#!/bin/sh
# RaspiBolt LND Mainnet: systemd unit for getpublicip.sh script
# /etc/systemd/system/20-raspibolt-welcome.sh

# make executable and copy script to /etc/update-motd.d/
# root must be able to execute bitcoin-cli and lncli

# set colors
color_red='\033[0;31m'
color_green='\033[0;32m'
color_yellow='\033[0;33m'
color_gray='\033[0;37m'

# set datadir
bitcoin_dir="/home/bitcoin/.bitcoin"
lnd_dir="/home/bitcoin/.lnd"

#Set to mount point of blockchain volume. This is used to calculate USB HDD usage %
ext_hdd="/mnt/hdd"

#Set to network device name (usually eth0 for ethernet or wlan0 for wireless on raspberry pi unless you have 'Predictable Network Names' turned on in raspi-config. To get network device name run ifconfig.)
network_name="eth0"

# get uptime & load
load=$(w | grep "load average:" | cut -c11-)

# get CPU temp
cpu=$(cat /sys/class/thermal/thermal_zone0/temp)
temp=$((cpu/1000))

# get memory
ram_avail=$(free -m | grep Mem | awk '{ print $7 }')
ram=$(printf "%sM / %sM" "${ram_avail}" "$(free -m | grep Mem | awk '{ print $2 }')")

if [ ${ram_avail} -lt 100 ]; then
  color_ram="${color_red}\e[7m"
else
  color_ram=${color_green}
fi

# get storage
sd_free_ratio=$(printf "%d" "$(df | grep "/$" | awk '{ print $4/$2*100 }')") 2>/dev/null
sd=$(printf "%s (%s%%)" "$(df -h | grep '/$' | awk '{ print $4 }')" "${sd_free_ratio}")
if [ ${sd_free_ratio} -lt 10 ]; then
  color_sd="${color_red}"
else
  color_sd=${color_green}
fi

hdd_free_ratio=$(printf "%d" "$(df | grep ${ext_hdd} | awk '{ print $4/$2*100 }')") 2>/dev/null
hdd=$(printf "%s (%s%%)" "$(df -h | grep ${ext_hdd} | awk '{ print $4 }')" "${hdd_free_ratio}")

if [ ${hdd_free_ratio} -lt 10 ]; then
  color_hdd="${color_red}\e[7m"
else
  color_hdd=${color_green}
fi

# get network traffic
network_rx=$(ifconfig ${network_name} | grep 'RX packets' | awk '{ print $6$7 }' | sed 's/[()]//g')
network_tx=$(ifconfig ${network_name} | grep 'TX packets' | awk '{ print $6$7 }' | sed 's/[()]//g')

# Bitcoin blockchain
btc_path=$(command -v bitcoin-cli)
if [ -n ${btc_path} ]; then
  btc_title="฿itcoin"
  chain="$(bitcoin-cli -datadir=${bitcoin_dir} getblockchaininfo | jq -r '.chain')"
[... 122 lines cut ...]
${color_red} ( : '~'.~.'~' : )   ${color_gray}Memory   ${color_ram}%-16s${color_gray}Sync    ${sync_color}%-14s${color_gray} ${alias_color}%s
${color_red}  ~ .~ (   ) ~. ~    ${color_gray}SSD      ${color_sd}%-16s${color_gray}        %-14s ${external_color}%s
${color_red}   (  : '~' :  )     ${color_gray}HDD      ${color_hdd}%-16s${color_gray}Public  ${public_color}%-14s ${color_gray}%s/%s Channels
${color_red}    '~ .~~~. ~'      ${color_gray}Traffic  ▲ %-12s  %-20s   ฿ %12s sat
${color_red}        '~'          ${color_gray}         ▼ %-12s  Mempool %-14s 🗲 %12s sat
${color_grey}                                              Connections %-10s ⌛%12s sat
${color_grey}                                                   In/Out %-10s ∑ %12s sat
${color_red}                     ${color_yellow}%s
%s %s
" \
"RaspiBolt" \
"-------------------------------------------------------------------" \
"${load}" "${temp}" \
"Resources free" "${btc_title}" "Lightning (LND)" \
"${ram}" "${sync}" "${ln_alias}" \
"${sd}" "${sync_behind}" "${ln_external}" \
"${hdd}" "${public}" "${ln_channels_online}" "${ln_channels_total}" \
"${network_tx}" "${public_addr}" "${ln_walletbalance}" \
"${network_rx}" "${mempool} tx" "${ln_channelbalance}" \
"$connections" "$ln_pendinglocal" \
"$inbound/$outbound" "$sum_balance" \
""

echo "$(tput -T xterm sgr0)"
```

Fillippone, 2019-02-18:
I tried to access my raspi remotely , and got a few errors running this script. It might be due to my errors, working through my mobile. I will let you know this evening. Did you check it yourself?

Fillippone, 2019-02-18:
I copied the Whole code in nano
saved and exited, then i followd stadicus guide

$ sudo mv /etc/update-motd.d /etc/update-motd.d.bak
>mv: cannot move '/etc/update-motd.d' to '/etc/update-motd.d.bak/update-motd.d': Directory not empty

$ sudo mkdir /etc/update-motd.d
>mkdir: cannot create directory ‘/etc/update-motd.d’: File exists

$ sudo cp 20-raspibolt-welcome /etc/update-motd.d/
$ sudo chmod +x /etc/update-motd.d/20-raspibolt-welcome
$ sudo ln -s /etc/update-motd.d/20-raspibolt-welcome /usr/local/bin/raspibolt
>ln: failed to create symbolic link '/usr/local/bin/raspibolt': File exists

As far as I understand this is expected as i already those command once. 

$ sudo raspibolt

i get the attached layout with an error regarding macaroons (i didn't get this error when running standard screen: 
[lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/readonly.macaroon: no such file or directory

layout
[https://imgur.com/a/w36f1As](url)

in addition to solving the macaroon problem I that  it would be better to move up the "PUBLIC" row in the bitcoin column and move also up all the lighting information on the right, erasing all the blank lines.

Fillippone, 2019-02-20:
any insight @robclark56 ?

Fillippone, 2019-02-21:
Another issue I noticed is a huge discrepancy between the traffic meter on the script and the one signalled by vnstat -d: can someone clarify the difference please? As you can see the node has been running since 4 days: vnstat would signal 20GiB data  while the script has gathered different data in that interva (less than 6 GiB).

 https://m.imgur.com/54hac62

robclark56, 2019-02-21:
1) What LND version are you running
 use: lncli with the --version argument

2) If you download the Stadicus file from here, does it run OK?
$ sudo apt-get install jq net-tools
$ cd /home/admin/download/
$ wget https://raw.githubusercontent.com/Stadicus/guides/master/raspibolt/resources/20-raspibolt-welcome -O test.sh
$ chmod +x test.sh
$ ./test.sh

Fillippone, 2019-02-21:
$ lncli with --version
>lncli version 0.5.1-beta commit=

$ sudo apt-get install jq net-tools
Reading package lists... Done
Building dependency tree
Reading state information... Done
jq is already the newest version (1.5+dfsg-1.3).
net-tools is already the newest version (1.60+git20161116.90da8a0-1).
0 upgraded, 0 newly installed, 0 to remove and 44 not upgraded.

Report working as attached image,  I run in that order:
1.vnstat -d
2. . updated  Stadicus script as per your instructions
3. your Script (macaroon error still present, symbols now readable (?), layout still to be adjusted)

[image]

Hope it helps.

robclark56, 2019-02-21:
Great info.

Now please run the script I gave you and send me a screen shot of the error

Rob

Fillippone, 2019-02-21:
This Is what I have done in the pic above. The last output following “sudo raspibolt” is the output of your script.

robclark56, 2019-02-21:
1) I notice that when you run test.sh (statdicus's unmodified code), there is no LND info. So that indicates a problem that has nothing to do with my mods. Is that because you are not running LND?

2) When you run my modified code, the error reported is that the macaroon_path file is not readable. Does the file reported in the error (i.e. /home/bitcoin......./mainnet/readonly.macaroon) exist?

3) Create this test file and run, and advise if you see an error
e.g.
 $ vi /tmp/temp.sh  <- enter text below
 $ chmod +x /tmp/temp.sh
 $ /tmp/temp.sh

#!/bin/sh
lnd_dir="/home/bitcoin/.lnd"
macaroon_path="${lnd_dir}/data/chain/bitcoin/mainnet/readonly.macaroon"
lncli="/usr/local/bin/lncli \
            --macaroonpath=${macaroon_path} \
            --tlscertpath=${lnd_dir}/tls.cert"
$lncli  getinfo

Fillippone, 2019-02-22:
Hello rob.
Sorry for the deal (Damn real life keeping me away from my bitcoins!)

1. Well spotted. Yes, I don't blame your script for that. My raspi is not funded, so i don't expect to see anything in the LND session. I just upgraded to latest LND 0.5.2 release, but  as I am running my node nehing two nested routers i know there are a few problems with this configuration, so I am leaving this out for the moment (btw, can you suggest a resource to know, if it is possbile to run a LND node behing two nested routers? as far as I know it is not possible due to UPnP not working in this configuration) 

2. I sorted this upgrading to last LND release. Now there's another error!

3 sorry instructuins are not clear to me. 
Can I use nano as I am not able to use vi?

Can you be more clear about what goes into the file and what must be run in command line? 

thanks,

robclark56, 2019-02-23:
I see my mods to display Bitcoin Connections is now working OK for you after you upgraded to V0.5. 
:-)
I wrote on V0.4.2, but edited for V0.5

Yes - you can use nano instead of vi (but as it is working ... you no longer need to create and run test,sh)

I will propose this change to Stadicus.

Layout: I tried to put the Bitcoin connections on one line but it was too long
  Connections xxxx (xxxx/xxxx)  
 
The 'blank' line between Sync OK & Public Yes is not really blank. Errors are displayed there if needed. (That is Stadicus' orig design and not my place to change)

I added the Pending Channels (hourglass) line in LND - which you will appreciate when you are opening & closing channels .. as your BTC seem to 'disappear' until the transaction competes. Note that closing channels  can take 1 week (by design) when you have to force-close a channel where the remote end has gone away.

Fillippone, 2019-02-23:
Ok I think we are done then.
I think this is a good improvement on the screen.
Thanks for the support.
I am going to close this issue.
Fillippone.

closed 2019-02-23 by Fillippone

Fillippone, 2019-03-01:
Hey @robclark56 could you post the PR to update the code?
Thanks,
Fillippone

robclark56, 2019-03-01:
One step at a time.

1) I have lodged a PR to fix a (separate) bug - related to calculating % free space on '/'
2) This weekend I am completing my migration to V0.5.2. When I have done that, I want to test my update, and then lodge a new PR

:-)

Fillippone, 2019-03-02:
No rush.
Thanks,
F

robclark56, 2019-03-06:
Pull Request done:
https://github.com/Stadicus/guides/pull/403