raspibolt/raspibolt #376: Zap Set-Up vs Public Key

xkra, 2019-02-10:
I use Linux and follow the step-by-step guide to set up the Zap Desktop Wallet.

I suspect that because I set up my Pi to allow logging in only with SSH keys (as described [here](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#login-with-ssh-keys)), I receive the following message when trying to [copy 'tls.cert' or the 'admin.macaroon' to my home directory](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_71_zap.md#on-your-linux-desktop-terminal).

`Permission denied (publickey).`

How can I resolve this issue?

Thanks in advance

Stadicus, 2019-02-12:
The two Linux commands work also with the SSH keys, as long as you authorized your pubkey also for the root user (see last part of the "[Login with SSH keys](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#login-with-ssh-keys)

```
$ sudo mkdir /root/.ssh
$ sudo cp /home/admin/.ssh/authorized_keys /root/.ssh/
$ sudo chown -R root:root /root/.ssh/
$ sudo chmod -R 700 /root/.ssh/
$ sudo systemctl restart ssh
```

xkra, 2019-02-12:
Thanks Stadicus,

Althoug I think I did that, meanwhile I actually seem to have crashed the old macbook pro with which I set up the SSH keys.
Lucky enough, I was about to send over the mainnet blockchain files to the HDD, which is why I just had the need for SSH keys turned off.

Is there a way to replace the old keys (also for root) with new ones from a new computer, this time running Windows? I am very hesitant to go trial&error as I did with all my other issues before...

Again many thanks in advance!

Stadicus, 2019-02-14:
You can put as many SSH keys in that authorized_keys file as you like, one per line. Also, if you are using a Raspberry Pi (not an Odroid HCx), you can always connect keyboard and screen and log in locally without SSH, then username / password work as well.

xkra, 2019-02-15:
Thanks, that was encouraging enough. Tried and successfully fixed it.

closed 2019-02-15 by xkra