raspibolt/raspibolt #377: updating to LND version 0.5.2

CryptoAstroNut, 2019-02-11:
With the new version 0.5.2 out, what commands need to be changed in the upgrade page to be able to do this? I tried doing it with the guide from the lnd page but its not working for me. I'm getting no such file or directory when I  enter the command `gpg --verify manifest-v0.5.2-beta.txt.sig`

CryptoAstroNut, 2019-02-11:
ok I took all the new update links and put in the samw order as the old update stuff.....before I run all this, How off am I? lol
`$ cd /home/admin/download
$  rm -f lnd-linux* manifest* pgp_keys.asc
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.5.2-beta/lnd-linux-armv7-v0.5.2-beta.tar.gz
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.5.2-beta/manifest-v0.5.2-beta.txt
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.5.2-beta/manifest-v0.5.2-beta.txt.sig 
$ wget  https://keybase.io/roasbeef/pgp_keys.asc | gpg --import 

$ sha256sum --check manifest-v0.5.2-beta.txt --ignore-missing
> lnd-linux-armv7-v0.5.2-beta.tar.gz: OK

$ gpg ./pgp_keys.asc
>  F8037E70C12C7A263C032508CE58F7F8E20FD9A2

$ gpg --import ./pgp_keys.asc
$ gpg --verify manifest-v0.5.2-beta.txt.sig 

>gpg: assuming signed data in 'manifest-v0.5.2-beta.txt'
>gpg: Signature made Thu Feb  7 13:29:16 2019 PST
>gpg:                using RSA key F8037E70C12C7A263C032508CE58F7F8E20FD9A2
>gpg: Good signature from "Olaoluwa Osuntokun <laolu32@gmail.com>" [ultimate]

$tar -xvzf vendor.tar.gz
$tar -xvzf lnd-source-v0.5.2-beta.tar.gz` 

$ sudo install -v -mod=vendor -ldflags "-X github.com/lightningnetwork/lnd/build.Commit=v0.5.2-beta"
$ sudo install -v -mod=vendor -ldflags "-X github.com/lightningnetwork/lnd/build.Commit=v0.5.2-beta" ./cmd/lncli 

$ lnd --version
> lnd version 0.05.2-beta

Stadicus, 2019-02-12:
So, everything worked? The guide should be updated, see this PR:
https://github.com/Stadicus/guides/pull/375

CryptoAstroNut, 2019-02-12:
Just saw you added the update. I haven't tried it yet. I was still looking at the update in the FAQ part of the troubleshooting and it only had the part for updating to 0.5.  The above was just me guessing what it would look like, I didn't want to enter it all and screw something up if it was wrong.
Ill try this and let you know. but as for all the other issues I had it's all corrected itself somehow and the node is up and running for now. Thanks.

CryptoAstroNut, 2019-02-12:
looks like everything downloaded ok but after restarting the system I'm getting a "failed to process consensus server notification"

CryptoAstroNut, 2019-02-13:
Seems like it resolved itself. Node is upgraded and running perfectly. Thank you. Ill close this now.

closed 2019-02-13 by CryptoAstroNut