raspibolt/raspibolt #379: Ext4 Alternatives

KayBeSee, 2019-02-12:
I started following this guide about a week ago and my blockchain is still only 35% of the way downloaded.

Before my raspberry pi came in the mail, I started downloading the blockchain to an external harddrive using Bitcoin QT. It took about a week for the whole chain to download. When I went to follow this guide, however, I reformatted my external hard drive (following the guide) and started downloading the blockchain again.

I think the reason it is taking so long is that the Pi itself doesn't have enough processing power to do it efficiently (I could be totally wrong, but that's just my hunch). So I am wondering if I can format the drive in a different way so that I can redownload it using my mac and then just plug it into the pi.

Stadicus, 2019-02-12:
Are you on Testnet or Mainnet? The Pi should have enough oomph to handle the Testnet chain, but it will take weeks for the mainnet.

This is why the guide is very clear on the following things (please correct me if something is unclear):

- Download the Bitcoin Mainchain on your regular computer to a regular harddisk (if external, then preferrably a USB3 disk)
- Transfer the Bitcoin Mainnet chain over your network with SCP (eg. WinSCP) to the Pi. 

In an earlier version, the guide included steps to use NTFS, but that leads to so many issues that I think using EXT4 & transferring the blockchain over the network is the safest route.

KayBeSee, 2019-02-12:
Ahh, I see that now.

Ok, this is very helpful. I'll just re-download it on my machine and then transfer it over the network.

One follow up question: If I wanted to send a friend a working node, is there an easy way to copy the full blockchain from an EXT4 drive on my pi to another EXT4 drive connected to the pi?

Stadicus, 2019-02-21:
That depends on your definition of "easy" :-)

If both devices are on the same network and can be accessed via SSH, then you can use `scp` from the command line to copy the data. See further information here:
https://linuxacademy.com/blog/linux/ssh-and-scp-howto-tips-tricks/

closed 2019-02-21 by Stadicus