raspibolt/raspibolt #383: Auto Unlock - Completes but does not unlock

ohenrik, 2019-02-15:
I'm trying to figure out why my unlock script is not working. The script seems to run fine and appends the unlock message to the audit.log file. However the lightning node does not unlock. It just sits there waiting to be unlocked.

Stadicus, 2019-02-17:
Any error message in the debug.log?

ohenrik, 2019-02-19:
No messages in debug.log :(

Stadicus, 2019-02-21:
Did you use the latest version of the unlock script?

daniel-vera-g, 2019-02-24:
> I'm trying to figure out why my unlock script is not working. The script seems to run fine and appends the unlock message to the audit.log file. However the lightning node does not unlock. It just sits there waiting to be unlocked.

It would be nice you could share your problem on #352 
Then we have everything in one place and do not have to switch between issues

Stadicus, 2019-03-01:
Yes, please check out the mentioned issue. Most issues arise from TLS problems.

closed 2020-02-28 by Stadicus