raspibolt/raspibolt #387: No RPC connection via public IP address

philippbuechel, 2019-02-19:
I seem not to be able to connect to my node from outside my network and I can't find what's the issue.

Wallets like "Union 7" or "Zap" are able to connect from within the network. 

The relevant part of lnd.conf looks like this:

> rpclisten=0.0.0.0:10009
>
> 
>tlsextraip=MY.FIXED.IP           # fixed ip is needed
>tlsextradomain=HOSTNAME.OF.IP	    # domain name

tls.cert and macaroon are copied into the wallet's preferences.

The router seems to be configured correctly as an `nmap -p10009 -Pn MY.FIXED.IP` shows the result 

>Starting Nmap 7.60 ( https://nmap.org ) at 2019-02-18 17:57 CET
>Nmap scan report for HOSTNAME.OF.IP (MY.FIXED.IP)
>Host is up (0.0041s latency).
>
>PORT      STATE SERVICE
>10009/tcp open  swdtp-sv
>
>Nmap done: 1 IP address (1 host up) scanned in 0.04 seconds

Nontheless, changing the IP from internal to external in the wallet's preferences prevents a connection. How can I troubleshoot this further?

Stadicus, 2019-02-21:
Did you adjust the subnet mask for the UFW rule?

philippbuechel, 2019-02-21:
Yes, UFW rules look like this

```
To                         Action      From
--                         ------      ----
22                         ALLOW       192.168.0.0/24             # allow SSH from local LAN
50002                      ALLOW       192.168.0.0/24             # allow Electrum from local LAN
9735                       ALLOW       Anywhere                   # allow Lightning
8333                       ALLOW       Anywhere                   # allow Bitcoin mainnet
18333                      ALLOW       Anywhere                   # allow Bitcoin testnet
22                         ALLOW       169.254.216.0/24           # allow SSH from local LAN
22                         ALLOW       192.168.10.0/24           
10009                      ALLOW       Anywhere                   # allow LND grpc from anywhere
9735 (v6)                  ALLOW       Anywhere (v6)              # allow Lightning
8333 (v6)                  ALLOW       Anywhere (v6)              # allow Bitcoin mainnet
18333 (v6)                 ALLOW       Anywhere (v6)              # allow Bitcoin testnet
10009 (v6)                 ALLOW       Anywhere (v6)              # allow LND grpc from anywhere

```
Tried to disable ufw, no success either way.

Stadicus, 2019-03-03:
If you use interface 0.0.0.0, UFW is not blocking anything and you're able to connect from within the network, I don't think it's the configuration of the node.

I'm sure you checked, but port forwarding is set correctly to the internal IP address of your node for port 10009?

AmazingDroopy, 2019-03-04:
The IP/host info is probably different in the tls.cert and that might prevent your app from connecting.

AmazingDroopy, 2019-03-04:
Check: https://ln-zap.github.io/zap-tutorials/iOS-remote-node-setup
In the last part the tutorial may give you clues why this is not working.

closed 2020-02-28 by Stadicus