raspibolt/raspibolt #389: LND will not start

pretyflaco, 2019-02-19:
Hello, 

My raspibolt has been running on mainnet for 2 months like a charm, until one day I received: 

```
$ lncli getinfo
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

$systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2019-02-19 23:51:20 CET; 1s ago
  Process: 1391 ExecStart=/usr/local/bin/lnd (code=exited, status=1/FAILURE)
 Main PID: 1391 (code=exited, status=1/FAILURE)
```
I have switched to user bitcoin and checked rights:

```
bitcoin@RaspiBOLT:~ $ ls -la /home/bitcoin/
total 48
drwxr-xr-x 6 bitcoin bitcoin 4096 Feb 19 22:30 .
drwxr-xr-x 5 root    root    4096 Dec  1 18:24 ..
-rw------- 1 bitcoin bitcoin 4141 Feb 19 23:47 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Dec  1 18:24 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Dec  1 18:24 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   16 Dec  1 20:59 .bitcoin -> /mnt/hdd/bitcoin
drwxr-xr-x 3 bitcoin bitcoin 4096 Jan  1 20:43 .cache
lrwxrwxrwx 1 bitcoin bitcoin   33 Jan  1 20:31 electrum-personal-server -> /mnt/hdd/electrum-personal-server
drwx------ 3 bitcoin bitcoin 4096 Jan  2 19:02 .gnupg
lrwxrwxrwx 1 bitcoin bitcoin   12 Dec  1 21:32 .lnd -> /mnt/hdd/lnd
drwx------ 6 bitcoin bitcoin 4096 Jan  1 20:43 .local
drwxr-xr-x 2 bitcoin bitcoin 4096 Dec  1 21:00 .nano
-rw-r--r-- 1 bitcoin bitcoin  675 Dec  1 18:24 .profile
-rw-r--r-- 1 bitcoin bitcoin    0 Feb 19 23:16 test
-rw-r--r-- 1 bitcoin bitcoin  259 Jan  1 20:32 .wget-hsts
```
```
bitcoin@RaspiBOLT:~ $ ls -la /home/bitcoin/.bitcoin/
total 21556
drwxr-xr-x 7 bitcoin bitcoin     4096 Feb 19 23:42 .
drwxr-xr-x 6 bitcoin bitcoin     4096 Jan  1 20:31 ..
-rw------- 1 bitcoin bitcoin       37 Feb 18 07:10 banlist.dat
-rw-r--r-- 1 bitcoin bitcoin      449 Dec 19 23:31 bitcoin.conf
-rw------- 1 bitcoin bitcoin        5 Feb 19 23:42 bitcoind.pid
drwx------ 3 bitcoin bitcoin    94208 Feb 19 08:18 blocks
drwxr-xr-x 2 bitcoin bitcoin    69632 Feb 19 23:54 chainstate
drwx------ 2 bitcoin bitcoin     4096 Feb 19 23:43 database
-rw------- 1 bitcoin bitcoin        0 Dec 19 23:33 db.log
-rw------- 1 bitcoin bitcoin  3852426 Feb 19 23:50 debug.log
-rw------- 1 bitcoin bitcoin   247985 Feb 19 23:41 fee_estimates.dat
drwx------ 3 bitcoin bitcoin     4096 Dec 19 23:34 indexes
-rw------- 1 bitcoin bitcoin        0 Dec 19 23:33 .lock
-rw------- 1 bitcoin bitcoin 10488986 Feb 19 23:41 mempool.dat
-rw------- 1 bitcoin bitcoin  4273474 Feb 19 23:41 peers.dat
-rw-r--r-- 1 bitcoin bitcoin        0 Feb 19 23:54 test
drwxr-xr-x 6 bitcoin bitcoin     4096 Dec 19 23:30 testnet3
-rw------- 1 bitcoin bitcoin  3002368 Feb 19 23:43 wallet.dat
-rw------- 1 bitcoin bitcoin        0 Dec 19 23:33 .walletlock
```
So far it looked alright, but then I noticed this: 

```
$ touch /home/bitcoin/.lnd/test
touch: cannot touch '/home/bitcoin/.lnd/test': Permission denied

bitcoin@RaspiBOLT:~ $ ls -la /home/bitcoin/.lnd/
ls: cannot access '/home/bitcoin/.lnd/..': Permission denied
ls: cannot access '/home/bitcoin/.lnd/tls.cert': Permission denied
ls: cannot access '/home/bitcoin/.lnd/.': Permission denied
ls: cannot access '/home/bitcoin/.lnd/logs': Permission denied
ls: cannot access '/home/bitcoin/.lnd/data': Permission denied
ls: cannot access '/home/bitcoin/.lnd/lnd.conf': Permission denied
ls: cannot access '/home/bitcoin/.lnd/tls.key': Permission denied
total 0
d????????? ? ? ? ?            ? .
d????????? ? ? ? ?            ? ..
d????????? ? ? ? ?            ? data
-????????? ? ? ? ?            ? lnd.conf
d????????? ? ? ? ?            ? logs
-????????? ? ? ? ?            ? tls.cert
-????????? ? ? ? ?            ? tls.key
```

If I understand correctly, I think user bitcoin should be allowed to access these files, right? How can I set permissions correctly here? Or is this unrelated to my issue?

I also tried starting LND manually after disabling lnd.service but it won't start - instead I get this: 

`bitcoin@RaspiBOLT:~ $ lnd
loadConfig: either bitcoin.active or litecoin.active must be set to 1 (true)`

but bitcoin.active=1 is already set in lnd.conf - thanks in advance for any help. 

Cheers,
k3m

Stadicus, 2019-02-21:
User "bitcoin" should be owner of the the whole hdd. Can you try:
`sudo chown -R bitcoin:bitcoin /mnt/hdd/*`

HelgeHunding, 2019-02-21:
I had the exact same error message. It started this morning when I wasn't able to connect to the Pi. I had to restart it manually, then it had to resync the last blocks of the Bitcoin blockchain. Once finished I could restart LND and it worked (but the Pi was really slow and commands took longer then usual). And when checking later LND was down and the blockchain was out of sync again. I´m in the process of syncing it up again, the weird part is that it´s currently at block height 563814 (two days ago) and the node´s been running fine for the past two days and was up a while today. Not sure what´s going on or if it might be related. If your blockchain in sync?
´bitcoin-cli getblockchaininfo´ and height

HelgeHunding, 2019-02-23:
I'm not 100% sure what caused my issue. But I think it was that I accidentally started bitcoind at the local ssd-drive when trying to fix another issue. Like `sudo start bitcoind`  that filled the memory of the Pi´s hard-drive and made it slow. That probably caused it to get out of sync with the network. So I had to cope the latest blocks, everything in blocks>index and the chainstate (no need to copy the blocks you've already copied once, compare the size to see if they're the same). Then everything started again, maybe unrelated to your issue but putting it out there if someone else experience the same.

pretyflaco, 2019-02-23:
> User "bitcoin" should be owner of the the whole hdd. Can you try:
> `sudo chown -R bitcoin:bitcoin /mnt/hdd/*`

I tried this - output changed, but still no permission

`admin ~ ฿ sudo chown -R bitcoin:bitcoin /mnt/hdd/*

[sudo] password for admin:
admin ~ ฿ touch /home/bitcoin/.lnd/test
touch: cannot touch '/home/bitcoin/.lnd/test': Permission denied
admin ~ ฿ ls -la /home/bitcoin/.lnd/
ls: cannot open directory '/home/bitcoin/.lnd/': Permission denied
`

pretyflaco, 2019-02-28:
> User "bitcoin" should be owner of the the whole hdd. Can you try:
> `sudo chown -R bitcoin:bitcoin /mnt/hdd/*`

How can I check whether command was successful and user bitcoin is owner of HDD? 
If it is with touch as in previous post and command failed, what could be the reason it failed?

Bitcoind is still running, so is Electrum Personal Server. But everything lnd related seems to have died :( 

```
admin ~ ฿ lncli getinfo
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
```
```
admin ~ ฿ systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-02-28 22:20:24 CET; 1s ago
  Process: 4293 ExecStart=/usr/local/bin/lnd (code=exited, status=1/FAILURE)
 Main PID: 4293 (code=exited, status=1/FAILURE)
```
```
admin ~ ฿ sudo journalctl -f -u lnd
-- Logs begin at Thu 2016-11-03 18:16:43 CET. --
Feb 28 22:20:24 RaspiBOLT systemd[1]: Started LND Lightning Daemon.
Feb 28 22:20:24 RaspiBOLT systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Feb 28 22:20:24 RaspiBOLT systemd[1]: lnd.service: Unit entered failed state.
Feb 28 22:20:24 RaspiBOLT systemd[1]: lnd.service: Failed with result 'exit-code'.
Feb 28 22:21:24 RaspiBOLT systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Feb 28 22:21:24 RaspiBOLT systemd[1]: Stopped LND Lightning Daemon.
Feb 28 22:21:24 RaspiBOLT systemd[1]: Started LND Lightning Daemon.
Feb 28 22:21:25 RaspiBOLT systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Feb 28 22:21:25 RaspiBOLT systemd[1]: lnd.service: Unit entered failed state.
Feb 28 22:21:25 RaspiBOLT systemd[1]: lnd.service: Failed with result 'exit-code'.
```

Stadicus, 2019-03-01:
Go to the root directory of the hdd and list its content:
```
$ cd /mnt/hdd
$ ls -la
```
the owner of the current directory (`.`) should be bitcoin, as well as all subdirectories. The owner of the parent directory (`..`) can be root, that's ok.

pretyflaco, 2019-03-04:
```
admin ~ ฿ cd /mnt/hdd
admin /mnt/hdd ฿ ls -la
total 1024040
drwxr-xr-x 6 bitcoin bitcoin       4096 Jan  1 20:31 .
drwxr-xr-x 3 root    root          4096 Dec  1 18:45 ..
drwxr-xr-x 7 bitcoin bitcoin       4096 Mar  3 22:23 bitcoin
drwxr-xr-x 3 bitcoin bitcoin       4096 Jan  5 16:28 electrum-personal-server
drw------- 4 bitcoin bitcoin       4096 Feb 28 23:17 lnd
drwx------ 2 bitcoin bitcoin      16384 Dec  1 18:30 lost+found
-rw------- 1 bitcoin bitcoin 1048576000 Dec  1 19:30 swapfile
```
looks as you described. weird.

pretyflaco, 2019-05-06:
Update: 
Since issue could not be resolved on Raspibolt and I had a very similar issue with LND on Thundroid, where updating to newest releases magically solved the issue, I decided to update all software on Raspibolt, too

Bitcoin Core Client to 0.18.0
`admin ~ ฿ bitcoind --version
Bitcoin Core Daemon version v0.18.0
`
bitcoind works fine without issues
Electrum Personal Server to 0.1.7
eps works fine without issues

Also ran:

`$ sudo apt-get update
$ sudo apt-get upgrade`
Successful

LND to 0.6 
`admin ~ ฿ lnd --version
lnd version 0.6.0-beta commit=v0.6-beta`
**Issue remains**

`bitcoin@RaspiBOLT:~ $ lnd
loadConfig: either bitcoin.active or litecoin.active must be set to 1 (true)
`

I went through whole of troubleshooting guide, unmounted hdd etc - the place where I get issues is that user bitcoin, despite being owner of and having writing rights on whole of /mnt/hdd cannot touch /home/bitcoin/.lnd/

`bitcoin@RaspiBOLT:~ $ touch /home/bitcoin/.lnd/test
touch: cannot touch '/home/bitcoin/.lnd/test': Permission denied

bitcoin@RaspiBOLT:~ $ ls -la /home/bitcoin/.lnd/
ls: cannot access '/home/bitcoin/.lnd/..': Permission denied
ls: cannot access '/home/bitcoin/.lnd/tls.cert': Permission denied
ls: cannot access '/home/bitcoin/.lnd/.': Permission denied
ls: cannot access '/home/bitcoin/.lnd/logs': Permission denied
ls: cannot access '/home/bitcoin/.lnd/data': Permission denied
ls: cannot access '/home/bitcoin/.lnd/test': Permission denied
ls: cannot access '/home/bitcoin/.lnd/lnd.conf': Permission denied
ls: cannot access '/home/bitcoin/.lnd/tls.key': Permission denied
total 0
d????????? ? ? ? ?            ? .
d????????? ? ? ? ?            ? ..
d????????? ? ? ? ?            ? data
-????????? ? ? ? ?            ? lnd.conf
d????????? ? ? ? ?            ? logs
-????????? ? ? ? ?            ? test
-????????? ? ? ? ?            ? tls.cert
-????????? ? ? ? ?            ? tls.key

`
Even user admin cannot touch nor cd to directory, when ls -la clearly says that directory is owned by bitcoin

`admin ~ ฿ touch /home/bitcoin/.lnd/test
touch: cannot touch '/home/bitcoin/.lnd/test': Permission denied

admin /mnt/hdd/lnd ฿ ls -la
ls: cannot open directory '.': Permission denied

admin ~ ฿ cd /home/bitcoin/.lnd/
-bash: cd: /home/bitcoin/.lnd/: Permission denied

admin ~ ฿ sudo su
root@RaspiBOLT:/home/admin# touch /home/bitcoin/.lnd/test
root@RaspiBOLT:/home/admin# sudo chown -R bitcoin:bitcoin /mnt/hdd/*
root@RaspiBOLT:~# cd /home/bitcoin/.lnd/
root@RaspiBOLT:/home/bitcoin/.lnd# ls -la
total 28
drw------- 4 bitcoin bitcoin 4096 May  7 01:00 .
drwxr-xr-x 6 bitcoin bitcoin 4096 Jan  1 20:31 ..
drwx------ 4 bitcoin bitcoin 4096 Dec  1 21:45 data
-rw-r--r-- 1 bitcoin bitcoin  372 May  6 23:31 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Dec  1 21:39 logs
-rw-r--r-- 1 bitcoin bitcoin    0 May  7 01:00 test
-rw-r--r-- 1 bitcoin bitcoin  794 Jan 19 18:17 tls.cert
-rw------- 1 bitcoin bitcoin  227 Jan 19 18:17 tls.key
root@RaspiBOLT:/home/bitcoin/.lnd# sudo su - bitcoin
bitcoin@RaspiBOLT:~ $ cd /home/bitcoin/.lnd/
-su: cd: /home/bitcoin/.lnd/: Permission denied
bitcoin@RaspiBOLT:~ $ exit
`

What could be the reason that user bitcoin is displayed as owner but cannot access /.lnd/ directory? I am not entirely sure that this is the reason for lnd not working, just trying to find a way forward how to fix this. Any hints appreciated. Thanks in advance.
cheers

Stadicus, 2020-02-28:
Closing as "too old" :-) Feel free to reopen if still relevant.

closed 2020-02-28 by Stadicus

d-hoffi, 2021-04-06:
Even if this thread is a bit "old". I got the same issue yesterday night. I did `sudo reboot` on my  raspibolt and got the same logs as @pretyflaco 
Like most things in LN I just gave it some time and after the night my lnd started as normal as usual. So the issue disappeared...