raspibolt/raspibolt #391: rpc incompatibility between auto-unlock and shango

m8tion, 2019-02-22:
My node works perfectly and I put it on tor.
I wanted to give a try on shango and I followed the guide. Fine, shango works when I'm on my wifi network and I find this app very usefull. But my node won't auto-unlock anymore (lnd gets stuck on unlocking when I reboot).
It seems there is an incompability in lnd.conf between the lines `rpclisten=0.0.0.0:10009     # for shango` and `restlisten=localhost:8080     # for auto-unlock`.
When I comment the line `rpclisten=0.0.0.0:10009` and reboot, auto-unlock works fine again (but no shango access of course).

Stadicus, 2019-03-01:
Have you tried putting both interfaces on `0.0.0.0`?

giulnz, 2019-03-20:
same problem, can't have both rpclisten=0.0.0.0:10009 and restlisten=localhost:8080 toghether works.
Also verify with 0.0.0.0 as suggested.

closed 2020-02-28 by Stadicus