raspibolt/raspibolt #395: Adding a blockchain explorer

danyR85, 2019-02-23:
First of all thank you and all the contributors for this awesome guide.

I would like to add a block explorer so as to be even less dependent on any 3rd parties.
I found several projects that allow this:

https://github.com/janoside/btc-rpc-explorer
https://github.com/Blockstream/esplora

My understanding is that it would require me to add a txindex=1 to bitcoin.conf and rescan the blockchain.

Is it time feasible to do so on the pi? 
would it interfere with electrum personal server?

Fillippone, 2019-02-23:
Reindexing the whole blockchain on the raspi took me more than one month (don’t extrapolate! Speed is decreasing).
Not sure about electrum personal server.

Stadicus, 2019-03-01:
You would need to add txindex=1 already when downloading the blockchain on your regular computer, reindexing on the Pi is really not feasible. But the Pi is able to keep up going forward. I have not tried any full block explorers (with their own additional index to link transactions) on the Pi, I think resources might be very tight... Better go for an Odroid or something similar.

Regarding EPS: it should not have any impact. But if you run eg. Esplora, you need to also run a full Electrum server (in this case Electrs, or a fork of it) and that eliminates the need for a "limited" personal electrum server. Just use the real thing. But again, probably not on a Pi.

danyR85, 2019-03-03:
Well I managed to reindex the blockchain and transfer it to the pi.
Meanwhile it works fine.

I am now trying to get the btc-rpc-explorer to work

badokun, 2019-04-17:
Did you get this to work?

OpenSystemCode, 2019-04-26:
My instructions to run: https://github.com/janoside/btc-rpc-explorer

I get some `Error: Error: ESOCKETTIMEDOUT` on a few tabs , don't know how to fix this. Hower this steps should point you in the right direction.

log into pi as admin user.

install nodejs:

`curl -sL https://deb.nodesource.com/setup_11.x | sudo bash -`
`sudo apt-get install -y nodejs `

Create new folder in the .home directory.

`mkdir btc-rpc-explorer`

move into it `cd btc-rpc-explorer/`

Clone the git repo:
`git clone https://github.com/janoside/btc-rpc-explorer.git .`

install project stuff :

`npm install`

Copy sample config:

`cp .env-sample .env`

edit it
nano .env

`#The active coin: BTC/LTC
BTCEXP_COIN=BTC
#Host/Port to bind to
BTCEXP_HOST=YOUR_PI_IP_ADDRESS
BTCEXP_PORT=3002

#Bitcoin RPC Credentials (URI -OR- HOST/PORT/USER/PASS)
BTCEXP_BITCOIND_URI=bitcoin://your_rpc_username:your_rpc_password@127.0.0.1:8332?timeout=10000
BTCEXP_BITCOIND_HOST=localhost
BTCEXP_BITCOIND_PORT=8332
BTCEXP_BITCOIND_USER=your_rpc_username
BTCEXP_BITCOIND_PASS=your_rpc_password
`
save it.

Create new SSH Session, login as admin. Now we create the firewall rule for Port 3002.

Change to root user:

`sudo su`

Rule:

`ufw allow from YOUR_LAN_NETWORK/24 to any port 3002 comment 'allow rpc-explorer local LAN'`

check it :
`ufw status
ufw enable
`
Go back to you first ssh session.

Start the explorer:

`node bin/cli.js`

If sucessfully some output will appear:

```
Loading env file: /home/admin/btc-rpc-explorer/.env
Running RPC Explorer for Bitcoin
Connecting via RPC to node at 127.0.0.1:8332
Connected via RPC to node. Basic info: version=170000, subversion=/Satoshi:0.17.0/, protocolversion=70015, services=000000000000040d
```
Open the explorer : YOUR_PI_IP_ADDRESS:3002

Stadicus, 2020-02-28:
It's happening! Feel free to already try it out: https://github.com/Stadicus/RaspiBolt/pull/547

danyR85, 2020-02-29:
works great!

closed 2020-03-31 by Stadicus