raspibolt/raspibolt #396: Port forwarding instead of UPnP for LND

dimarkov, 2019-02-24:
Hi, 

could someone clarify here why LND requires UPnP? Why cannot we just make port forwarding rules for LND?

Following the Stadicus's guide for setting ufw we have blocked external outgoing/incoming on all ports except 9735 and 8333. Why it is not enough to define port forwarding rules for 9735 as we already have define forwarding for 8333?

I would prefer to have the UPnP disabled on my router as it is a security issue. However, I do not have a static external ip, which would be another way to make my node public ...

compane, 2019-02-24:
It depends on your internet connection. 

static public IP: UPnP not necessary
dynamic public IP: UPnP or [script ](https://github.com/robclark56/RaspiBolt-Extras/blob/master/RB_extra_02.md) necessary

dimarkov, 2019-02-24:
Thanks!

Just to clarify, if I set up port forwarding for 9735 and restart lnd whenever the external IP address changes I do not need to enable UPnP. Is that correct?

Is it possible to loose open channels upon restart of lnd? Do you know how is the change of IP address affecting open channels?

Stadicus, 2019-03-01:
In general, channels are reestablished quite reliably, even with one party changing its address constantly. Unless both parties announce wrong ip addresses, I don't think that's a topic to worry about.

closed 2019-03-03 by dimarkov