raspibolt/raspibolt #397: Bitcoind stuck on startup

Laloown, 2019-02-24:
Im trying to make Stadicus guide work for both Bitcoin Core & LND

I did download the blockchain and moved to the HDD on my Rasp3

However, when starting the bitcoind service, im having this debug.log

`
2019-02-23T23:47:58Z Bitcoin Core version v0.17.1 (release build)
2019-02-23T23:47:58Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-02-23T23:47:58Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-02-23T23:47:58Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-02-23T23:47:58Z Using the 'standard' SHA256 implementation
2019-02-23T23:47:58Z Default data directory /home/bitcoin/.bitcoin
2019-02-23T23:47:58Z Using data directory /home/bitcoin/.bitcoin
2019-02-23T23:47:58Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-02-23T23:47:58Z Using at most 40 automatic connections (1024 file descriptors available)
2019-02-23T23:47:58Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-02-23T23:47:58Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-02-23T23:47:58Z Using 4 threads for script verification
2019-02-23T23:47:58Z scheduler thread start
2019-02-23T23:47:58Z HTTP: creating work queue of depth 16
2019-02-23T23:47:58Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-02-23T23:47:58Z HTTP: starting 4 worker threads
2019-02-23T23:47:58Z Using wallet directory /home/bitcoin/.bitcoin
2019-02-23T23:47:58Z init message: Verifying wallet(s)...
2019-02-23T23:47:58Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2019-02-23T23:47:58Z Using wallet wallet.dat
2019-02-23T23:47:58Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.bitcoin/database ErrorFile=/home/bitcoin/.bitcoin/db.log
2019-02-23T23:47:58Z Cache configuration:
2019-02-23T23:47:58Z * Using 2.0MiB for block index database
2019-02-23T23:47:58Z * Using 8.0MiB for chain state database
2019-02-23T23:47:58Z * Using 90.0MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-02-23T23:47:58Z init message: Loading block index...
2019-02-23T23:47:58Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2019-02-23T23:47:58Z Opened LevelDB successfully
2019-02-23T23:47:58Z Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
`

Same log goes on and on every minute.

Tried with bitcoind -reindex, however, all the reindex blocks goes to my /root/.bitcoin/blocks folder and gets filled in 2 hours (12gb of reindex) I also don't understand why the -reindex option goes towards to my /root/ folder instead of /mnt/hdd when i try to run bitcoind normally.

Anyone has any clue about this? Thanks you!

An even bigger loop log file

`2019-02-24T14:19:56Z Bitcoin Core version v0.17.1 (release build)
2019-02-24T14:19:56Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-02-24T14:19:56Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-02-24T14:19:56Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-02-24T14:19:56Z Using the 'standard' SHA256 implementation
2019-02-24T14:19:56Z Default data directory /home/bitcoin/.bitcoin
2019-02-24T14:19:56Z Using data directory /home/bitcoin/.bitcoin
2019-02-24T14:19:56Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-02-24T14:19:56Z Using at most 40 automatic connections (1024 file descriptors available)
2019-02-24T14:19:56Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-02-24T14:19:56Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-02-24T14:19:56Z Using 4 threads for script verification
2019-02-24T14:19:56Z scheduler thread start
2019-02-24T14:19:56Z HTTP: creating work queue of depth 16
2019-02-24T14:19:56Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-02-24T14:19:56Z HTTP: starting 4 worker threads
2019-02-24T14:19:56Z Using wallet directory /home/bitcoin/.bitcoin
2019-02-24T14:19:56Z init message: Verifying wallet(s)...
2019-02-24T14:19:56Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2019-02-24T14:19:56Z Using wallet wallet.dat
2019-02-24T14:19:56Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.bitcoin/database ErrorFile=/home/bitcoin/.bitcoin/db.log
2019-02-24T14:19:57Z Cache configuration:
2019-02-24T14:19:57Z * Using 2.0MiB for block index database
2019-02-24T14:19:57Z * Using 8.0MiB for chain state database
[... 95 lines cut ...]
2019-02-24T14:24:28Z Using at most 40 automatic connections (1024 file descriptors available)
2019-02-24T14:24:28Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-02-24T14:24:28Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-02-24T14:24:28Z Using 4 threads for script verification
2019-02-24T14:24:28Z scheduler thread start
2019-02-24T14:24:28Z HTTP: creating work queue of depth 16
2019-02-24T14:24:28Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-02-24T14:24:28Z HTTP: starting 4 worker threads
2019-02-24T14:24:28Z Using wallet directory /home/bitcoin/.bitcoin
2019-02-24T14:24:28Z init message: Verifying wallet(s)...
2019-02-24T14:24:28Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2019-02-24T14:24:28Z Using wallet wallet.dat
2019-02-24T14:24:28Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.bitcoin/database ErrorFile=/home/bitcoin/.bitcoin/db.log
2019-02-24T14:24:29Z Cache configuration:
2019-02-24T14:24:29Z * Using 2.0MiB for block index database
2019-02-24T14:24:29Z * Using 8.0MiB for chain state database
2019-02-24T14:24:29Z * Using 90.0MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-02-24T14:24:29Z init message: Loading block index...
2019-02-24T14:24:29Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2019-02-24T14:24:33Z Opened LevelDB successfully
2019-02-24T14:24:33Z Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
2019-02-24T14:24:55Z LoadBlockIndexDB: last block file = 1409
2019-02-24T14:24:55Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=120, size=133788575, heights=547036...547155, time=2018-10-23...2018-10-24)
2019-02-24T14:24:55Z Checking all blk files are present...
2019-02-24T14:24:56Z Opening LevelDB in /home/bitcoin/.bitcoin/chainstate`

lavenetz, 2019-03-21:
Hi,
having a similar problem: Part with Testnet3 was OK for Bitcoin and LND. Blockchain for mainnet copied with cp -a by mounting an ext4 usb drive. Started bitcoind with "$ sudo systemctl start bitcoind" without any message, but command "$ systemctl status bitcoind.service" is hanging:

admin@RaspiBolt:~ $ sudo systemctl start bitcoind
[sudo] password for admin: 

admin@RaspiBolt:~ $ systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-03-21 12:48:06 CET; 28s ago
  Process: 13902 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.p
  Process: 13896 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)
 Main PID: 13911 (code=exited, status=1/FAILURE)

admin@RaspiBolt:~ $ sudo tail -f /home/bitcoin/.bitcoin/debug.log
[sudo] password for admin: 

2019-03-21T11:03:57Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-03-21T11:03:57Z Using 4 threads for script verification
2019-03-21T11:03:58Z scheduler thread start
2019-03-21T11:03:58Z Binding RPC on address ::1 port 8332 failed.
2019-03-21T11:03:58Z Binding RPC on address 127.0.0.1 port 8332 failed.
2019-03-21T11:03:58Z Unable to bind any endpoint for RPC server
2019-03-21T11:03:58Z Error: Unable to start HTTP server. See debug log for details.
2019-03-21T11:03:58Z Shutdown: In progress...
2019-03-21T11:03:58Z scheduler thread interrupt
2019-03-21T11:03:58Z Shutdown: done
2019-03-21T11:05:02Z 

2019-03-21T11:05:02Z Bitcoin Core version v0.17.1 (release build)
2019-03-21T11:05:02Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-03-21T11:05:02Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-03-21T11:05:02Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-03-21T11:05:02Z Using the 'standard' SHA256 implementation
2019-03-21T11:05:02Z Default data directory /home/bitcoin/.bitcoin
2019-03-21T11:05:02Z Using data directory /home/bitcoin/.bitcoin
2019-03-21T11:05:02Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-03-21T11:05:02Z Using at most 40 automatic connections (1024 file descriptors available)
2019-03-21T11:05:02Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-03-21T11:05:03Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-03-21T11:05:03Z Using 4 threads for script verification
2019-03-21T11:05:03Z scheduler thread start
2019-03-21T11:05:04Z Binding RPC on address ::1 port 8332 failed.
2019-03-21T11:05:04Z Binding RPC on address 127.0.0.1 port 8332 failed.
2019-03-21T11:05:04Z Unable to bind any endpoint for RPC server
2019-03-21T11:05:04Z Error: Unable to start HTTP server. See debug log for details.
2019-03-21T11:05:04Z Shutdown: In progress...
2019-03-21T11:05:04Z scheduler thread interrupt
2019-03-21T11:05:04Z Shutdown: done
2019-03-21T11:06:06Z 
...

Also tried with bitcoind -reindex-chainstate, however, all the blocks go to /root/.bitcoin/blocks folder and gets filled within 18 hours already (8 GB) instead of /mnt/hdd.

Anyone has any clue about this? Thanks you!

MrPaz, 2019-05-02:
Wow, I'm having the same issue too.  Daemon is getting stuck at exact same point... `Using obfuscation key for /home/bitcoin/.bitcoin/chainstate: ` Doesn't give any shut down messages, it just stalls for several minutes then restarts and never gets past the obfuscation key message.  

For me it started a week ago, after a power outage. I read somewhere to reindex-chainstate, so tried reindexing on laptop and moved entire chain data onto the pi hdd again and still having his problem.  Re-flashed a new sd card.  No dice.  Someone suggested it may be corrupted hardware.  But idk cuz I can still read and write to hdd seemingly ok.

@lavenetz  , I think you need to start your daemon using user bitcoin to use the .bitcoin folder which points to hdd.  using admin i don't think there's a symbolic link to the hdd

closed 2020-02-28 by Stadicus