raspibolt/raspibolt #400: Tor with Bitcoind

dschoel, 2019-02-28:
I have had bitcoind up and running without Tor successfully.
I installed Tor, per the instructions, and the startup of bitcoind shows the appropriate entries in debug.log for Tor:

`2019-02-28T21:59:00Z` tor: Got service ID c3pq7qovpen4xpij, advertising service {XXXX}.onion:8333`

However, my .onion address won't show up on bitnodes.earn.com and I'm getting the following errors in my log:

`2019-02-28T22:10:38Z Socks5() connect to 149.233.160.130:8333 failed: general failure`

What should I check in my firewall and router to resolve?
Or am I missing something else?

daniel-vera-g, 2019-04-12:
I've the same issue.  
I followed the Tor tutorial in the bonus section.

---

Here is my desription of th issue::

Till now, my node was public, had more than 8 connections(=some inbound connections?) and checking for the port 8333 it was accessible for the outside.

For tor, I have this config:

Torrc:
```
ControlPort 9051
CookieAuthentication 1
CookieAuthFileGroupReadable 1
```

bitcoin.conf:
```
# Bitcoind options
server=1
daemon=1
txindex=1
...

dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

# Tor config
proxy=127.0.0.1:9050
bind=127.0.0.1
listenonion=1
```

In the bitcoind debug log everything seems good(Tor starts and the service ID gets shown), `bitcoin-cli getnetworkinfo` also shows that everything works and the node quickly gains peers.

Nevertheless, it stops when it gets 8 peers(outbound connections) and it is no more recognizable from the outside(port 8333).

I want to have a node, that uses tor and is recognizable from the outside(Inbound connections). Currently apparently only the first assumption is true.
I'm wondering, if I've forgot something.

---

I also tried both methods described in the [wiki](https://en.bitcoin.it/wiki/Setting_up_a_Tor_hidden_service)

daniel-vera-g, 2019-04-14:
> I have had bitcoind up and running without Tor successfully.
> I installed Tor, per the instructions, and the startup of bitcoind shows the appropriate entries in debug.log for Tor: [...]

1. What tor version are you running?
2. What's the configuration for Tor and Bitcoin in your `.torrc` & `bitcoin.conf` files? 
3. How many peers do you have(`bitcoin-cli getconnectioncount`)?

> However, my .onion address won't show up on bitnodes.earn.com

Mine does not show either, but I've more than 8 peers. That means, that I am reachable. So it seems to be ok.

> 2019-02-28T22:10:38Z Socks5() connect to 149.233.160.130:8333 failed: general failure

I think that this means there is some error with the server you want to connect to(`149.233.160.130:8333`). I am not sure if this is your or their fault. If bitcoind works and you've more than 8 peers, I would say that's ok and it could be `149.233.160.130:8333`fault.

> What should I check in my firewall and router to resolve?

If bitcoind works and it gains peers, it should not be needed.

closed 2020-02-28 by Stadicus