raspibolt/raspibolt #401: System overvie

Lit3block, 2019-03-01:
Hi guys
I've a little problem with my system overview.

despite I run bitcoind and lnd over tor, the system overview show me my real ip (not .onion address). 
moreover the node is shown as not reachable, but the commands bitcoin-cli getnetworkinfo, getpeerinfo and getblockchaininfo tell me that everything is ok and the node is online and reachable.

How can I fix that?

[image]

Stadicus, 2019-03-03:
The login script is not really maintained anymore. Maybe we can "borrow" the script from the RaspiBlitz project? PRs welcome!

https://github.com/rootzoll/raspiblitz/blob/master/home.admin/00infoBlitz.sh

Fillippone, 2019-03-09:
Why did you state the system overview is not maintained anymore?
There is also a pull request from @robclark56

daniel-vera-g, 2019-04-14:
> Hi guys
> I've a little problem with my system overview.
>
> despite I run bitcoind and lnd over tor, the system overview show me my real ip (not .onion address).
> moreover the node is shown as not [...]

I cannot confirm that it shows the ip. I am also running tor and it shows my onion adress.

The script makes an request to bitnodes.com to see if your node is reachable. I don't know why, but bitnodes still does not show my node as reachable nevertheless it has more than 8 connections. So, I would not mind.

BlackBullTradingPost, 2019-06-22:
> Hi guys
> I've a little problem with my system overview.
>
> despite I run bitcoind and lnd over tor, the system overview show me my real ip (not .onion address).
> moreover the node is shown as not [...]

The script still works, but when applying the bitcoin core release 18 you'll have to modify your bitcoin.conf file to include the following line:

#Connection Settings
**port=8333** 
Save, then restart bitcoind 
$ sudo systemctl restart bitcoind

Septem151, 2019-12-23:
> The script still works, but when applying the bitcoin core release 18 you'll have to modify your bitcoin.conf file to include the following line:
> 
> #Connection Settings
> **port=8333**
> Save, then restart bitcoind
> $ sudo systemctl restart bitcoind

Does this fix apply to release v19.0.1?

Stadicus, 2020-02-28:
I'll probably remove all code that calls external systems in the near future. The "public reachable" info is quite unreliable anyway, and redundant with the "Connections In" count.

Longmont, 2020-12-19:
I'm a little confused. I can make connections and open channels with others. I need more liquidity though. My Bitcoin node has peers. Yet I was told that my node being "Public Not Reachable" is a problem. And this is the root of why creating liquidity on lnd is a problem. That's illogical to me. Also my node shows on 1ml, and I did create an onion address as well. Bottom line, the code line Public: Not Reachable is unnecessary? I believe Stadicus referenced above as well, it's not necessary. Due to the IP being listed below the Bitcoin banner and version. I'm of the opinion that line of code should be hashed out/removed going forward. Be merciful, I'm still a noob to this Bitcoin world. Many things left to learn.

closed 2021-12-17 by github-actions[bot]