raspibolt/raspibolt #402: Steps to connect ZAP-IOS to your Raspibolt...

AmazingDroopy, 2019-03-04:
Follow this steps to connect the ZAP-IOS wallet to you RaspiBolt...

# First use the same access settings as the Shango wallet...
# Tell LND to listen for gRPC communication. Add the following line to your lnd configuration file in the section [Application Options]

$ sudo nano /home/bitcoin/.lnd/lnd.conf
   rpclisten=0.0.0.0:10009

# Open port 10009 so that ZAP-IOS wallet can talk to your Lightning node. 
# Modify UFW to only from within the local network (The line sudo ufw allow from 192.168.0.0/24… below assumes that the IP address of your Pi is something like 192.168.0.???, the ??? being any number from 0 to 255. If, for example, your IP address is 12.34.56.78, you must adapt this line to sudo ufw allow from 12.34.56.0/24…., see more details)

$ sudo ufw allow from 192.168.0.0/24 to any port 10009 comment 'allow LND grpc from local LAN'
$ sudo ufw enable
$ sudo ufw status
Restart LND and unlock wallet

$ sudo systemctl restart lnd
$ lncli unlock

# Install Go on Raspberry Pi….
# Update your Pi...
$ sudo apt-get update
$ sudo apt-get upgrade

# download Go binaries…
$ wget https://dl.google.com/go/go1.11.linux-armv6l.tar.gz

# unpack Go binaries…
$ sudo tar -xvf go1.11.linux-armv6l.tar.gz

# copy Go to env folder…
$ sudo mv go /usr/local

# Setup environment variables…(for this session only)
$ export GOROOT=/usr/local/go
$ export GOPATH=$HOME/gocode
$ export PATH=$GOPATH/bin:$GOROOT/bin:$PATH

# Install LND Connect..
# go get -d github.com/LN-Zap/lndconnect
$ cd $GOPATH/src/github.com/LN-Zap/lndconnect
$ make

# run LNDConnect with path to macaroon…
$ lndconnect --lnddir=/home/bitcoin/.lnd --adminmacaroonpath=/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon

# Scan QR code with ZAP wallet on IOS …
Scan the QR code and check the IP address you want to use to connect. 

Have fun.

ZAP-IOS supports to switch between multiple connected LND Nodes.

Stadicus, 2019-03-05:
Would you like to create a proper pull request?

AmazingDroopy, 2019-03-05:
Apologies, I realized too late this should be a PR.
I will do as requested.

Richard.

bckbmath, 2019-03-31:
I cant get zap to connect after following these instructions, is there something missing?

bckbmath, 2019-03-31:
"Unable to connect to host: Failed to connect before the deadline"

AmazingDroopy, 2019-04-01:
Is your IOS device able to ping the LND host? The setup assumes you connect via private network, not via Internet.

Does your LND host allow connections? (Check lnd.conf)

Did you change the IP in the LNconnect phase in ZAP mobile so it reflects your internal accessible IP? I had to, because the QR gives me the external public IP, which is not open for connections.

Richard.

bckbmath, 2019-04-01:
No I can't ping the LND host, not sure why.

This is my lnd.conf - i think it allows connections?

debuglevel=debug
maxpendingchannels=5

color=#68F442
rpclisten=0.0.0.0:10009

# Your router must support and enable UPnP, otherwise delete this line
nat=true

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

bckbmath, 2019-04-01:
Changing the ip to a local one worked a treat cheers

AmazingDroopy, 2019-04-01:
Great,

 I think that changing IP is mentioned in the guide. Maybe not very explicit.

Richard.

Stadicus, 2019-04-18:
There is now a proper bonus guide, based on @AmazingDroopy 's initial contribution. 
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_72_zap-ios.md

closed 2019-04-18 by Stadicus