raspibolt/raspibolt #404: Wladimir J. van der Laan's PGP key is now expired

Barno, 2019-03-07:
In guides/raspibolt/raspibolt_30_bitcoin.md 

when I check the signature with 
`gpg --verify SHA256SUMS.asc`

I get this warning
```
gpg: Signature made Tue Dec 25 08:03:05 2018 UTC
gpg:                using RSA key 90C8019E36C2E964
gpg: Good signature from "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" [expired]
gpg: Note: This key has expired!
Primary key fingerprint: 01EA 5486 DE18 A882 D4C2  6845 90C8 019E 36C2 E964
```
I have to update the keys ([here](http://www.erisian.com.au/bitcoin-core-dev/log-2019-02-11.html), line 192. --refresh-keys.)

Then I installed
```
sudo apt-get install dirmngr
```
And updated the keys
```
gpg --refresh-keys laanwj@gmail.com
```

After that I can run again the check
` gpg --verify SHA256SUMS.asc`
And now it works

Stadicus, 2019-03-13:
Opened an official issue on bitcoin/bitcoin
https://github.com/bitcoin/bitcoin/issues/15592

Stadicus, 2019-03-16:
The updated key has been merged with bitcoin.org, bitcoincore.org will probably follow shortly.

closed 2019-03-16 by Stadicus