raspibolt/raspibolt #407: Mac OS Guide?

LoganJastremski, 2019-03-09:
Hello!

Excellent guide! Overall this guide assumes that you will use a Windows machine for tasks and command prompt. In the future would you be willing to expand on this guide for Mac OS & terminal users?

Thanks!

Stadicus, 2019-03-13:
Thanks! The guide tries to be agnostic to the OS of your regular computer, but it's undeniably written from a Windows perspective. Where Mac OS differs, we tried to include the relevant information, eg. here:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#login-with-ssh-keys

Feel free to point out specific steps where you would like to have additional information for MacOS users, or create a pull request.

LoganJastremski, 2019-03-23:
The first problem that I'm encountering is with syncing the blockchain. Macs inherently don't support formatting for Ext4 file system unless using paid 3rd-party software to my knowledge. Are other formats supported or that you would recommend?

An additional problem that I encountered was that my computer doesn't have enough storage to pre-sync the chain to the internal hard drive - the workaround was to format the external hard drive with the paid 3rd party software to Ext4 then sync the drive, but I was still encountering problems with syncing there as well. Any suggestions? Once again, thanks for the excellent guide!

Stadicus, 2019-04-18:
Hej @LoganJastremski, it's hard to put the blockchain data directly onto the drive designated for the Pi, there were tons of issues in the past, this is why I don't use that method. 

If you don't have enough storage elsewhere, I guess that's the only option. But you coud still use a second external drive, maybe borrow one from a friend?

anonymoustrident, 2019-04-27:
@LoganJastremski , I did follow the guide all through using a Mac Machine. I also bought a HDD just for this purpose and change the default sync directory to the external HDD instead. Works well

closed 2020-02-28 by Stadicus