raspibolt/raspibolt #410: How to get a new URI (pubkey), when switching from clearnet to Tor?

raindogdance, 2019-03-18:
I set up Tor as described in the guide and now `lncli getinfo` shows two URIs, both with the same pubkey, but different endings (one with my clearnet IP and one with the onion address). 

Currently I still have channels opened on my clearnet URI. I would like to set up my node with a completely new pubkey/URI, which is not linked to the previous one (which at the moment appears along with my clearnet IP on 1ml.com).  

The guide says: `it is recommended to close all Lightning channels and start a brand new node on Tor.`.  Closing channels is straight forward, but I am not sure how exactly to set up a new node? Do I back up my wallet and then simply delete everything in the Lightning folders and follow the guide from scratch for the Lightning part?  

And will it guarantee a new pubkey being generated?  

The search gave out this issue thread: https://github.com/Stadicus/guides/issues/335, where the same question on the pubkey was posted, but no answers yet.

giulnz, 2019-03-21:
Before try anything i suggest you to close channels and move founds.
As i understand Easy way to do what you want is stop lnd.service (sudo systemctl stop lnd) and remove .lnd folder. 
when you restart lnd it will recreate the file, so before make the dir .lnd and create the right lnd.conf file with tor setting.
Remember also to copy tls.cert and macaroon to admin/.lnd folder

raindogdance, 2019-03-28:
Thank you!

This sounds somewhat easy on paper but probably less so in practice :)  

Let's see if I can do it step by step.  

So I managed to copy tls.cert and macaroon.db into my `/home/admin/downloads` folder, so those two files won't get deleted.  

First, I will close the channels, wait for confirmations, move all of the funds from the LND wallet to another wallet, and stop the LND client.  

Next, I'll need to delete the .lnd folder. Since that's only the symbolic link(?), is it enough if I do `rm -rf .lnd`, or will I need to go to `/mnt/hdd/lnd` and delete the contents of the `lnd` folder (since that's the location to which the symbolic link points)?  

After that, I simply will follow the guide again, starting from this step: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#lightning-lnd  

After re-installing and following the appropriate guide for the TOR part, LND should be running on TOR completely (with a completely new URI address)?

Did I get everything right? Just checking before starting tinkering :)

giulnz, 2019-03-28:
You don't need to reinstall lnd and only to "restore" lnd.conf if is your right lnd configuration .. normally it's the default file as in stadicus guide that you linked + extra lines in stadicus tor guide. 
I know it works like this, i don't know what will happen if you restore macaroon.db and tls.cert but i don't think it's a good move. 

0-Backup your lnd.conf   
  As admin user
  $ mkdir /home/admin/backup 
  $ sudo cp /home/bitcoin/.lnd/lnd.conf /home/admin/download
1-close channels
2-move founds in a your safe address
3-stop lnd   
  $ sudo systemctl stop lnd
4-remove .lnd   
  $ sudo rm -r /home/bitcoin/.lnd 
5-create .lnd dir and conf file   
  as bitcoin user
  $ mkdir /home/bitcoin/.lnd  
  $ nano /home/bitcoin/.lnd/lnd.conf
(copy the file from your old or make as https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#lightning-lnd 
plus tor extra line)
6- continue to follow https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md#lightning-lnd 
from "RUN LND" 
you don't need to do "Autostart LND" section. 
7- maybe you need to start lnd if is not running  as user $ sudo systemctl start lnd

giulnz, 2019-03-28:
probably you can close channels and don't move the found, just restore the same seed (the 24 words plus passphrase) when you create the new wallet, i'm not an expert i'm just try to help so please be careful if you are in mainnet.

giulnz, 2019-03-28:
uri is derived from seed so better use another one if your previous uri are linked to your ip

raindogdance, 2019-03-29:
Thank you for the step by step explanation! :)  
Currently waiting for some of my channels to get active again, so I can close them normally (without "force"), and then I'll go through the re-installing process. Will report back if everything went fine. 

Probably those steps should be a part of the guide, as many people will check out the TOR functionality only after they set up everything? @Stadicus what do you think?  

> uri is derived from seed so better use another one if your previous uri are linked to your ip  

Excellent info, btw, thanks again :)

Stadicus, 2019-04-02:
Thanks for the great explanation @giulnz, much appreciated!

Stadicus, 2019-04-18:
Pull requests welcome... :-)

raindogdance, 2019-05-19:
So the good news is that I managed to get a new pubkey, after all :)  

In addition to the steps described by @giulnz, I also deleted the whole `/mnt/hdd/lnd` folder, as well as the `lncli` and `lnd` files in `/usr/local/bin` folder.  And re-installed LND as described by the guide. 

As a side note, I observed that importing my old seed gives out a new pubkey as well, so the old seed doesn't seem to produce the same old pubkey (at least, if you delete and re-install everything)?  

Now, the bad news is that I cannot get LND run through Tor anymore, apparently. `lncli getinfo` shows the URI with my clearnet IP.  

I wiped and re-installed LND twice (once with creating a whole new seed, and once trying out my new one, just to check if the pubkey would change or not) by following the setup guide and the TOR bonus section for LND. Double-checked the `/home/bitcoin/.lnd/lnd.conf` as well, but everything seems as it should be. Restarted LND a few times as well.   

So out of ideas for the moment.

closed 2020-02-28 by Stadicus