raspibolt/raspibolt #411: Raspberry fails booting when no HDD is attached

Pythoniasm, 2019-03-19:
I am running the up-to-date configuration of raspibolt as well as the raspibolt welcome screen. 

When the HDD is not attached/turned on during the booting, the raspberry won't let me log in via SSH and shutsdown shortly after.

Is there a way to give it a fall back if the HDD is not mounted?

spookiestevie, 2019-04-07:
Hi, i solved this by going into the /ect/fstab file and changing some options...
I think you want the -nofail

Original:
UUID=123456 /mnt/hdd ext4 noexec,defaults 0 0
New:
UUID=123456 /mnt/hdd ext4 noexec,nofail,defaults 0 0

Stadicus, 2019-04-18:
Thanks @spookiestevie for the reminder, I wanted to change that for some time now... All options are now explicitly set, including filesystem check on boot:
https://github.com/Stadicus/guides/commit/da34ba26cfbe5ff48d5cb241ce2af94292157bd1

closed 2019-04-18 by Stadicus