raspibolt/raspibolt #415: Copying Bitcoin Synced Node Over To Pi HDD

anonymoustrident, 2019-03-21:
Have been following and reading through all closed and currently open issues to address this. I am sure it must have been written somewhere but I can't find it. 

I am now at the TAB of Bitcoin after securing the whole PI ready for the BTC node . 
1. I have a fully synced node on my local machine 
I can't find any detail instruction on the guide as to how you transfer the chaindata over to the HDD we have set up earlier? I saw WINSCP briefly but it is for windows , So for Mac, we need to SSH the chaindata over if we do not wish to unmount the HDD , mount it to the local and transfer the files into the EXT4 HDD

2. wherein the HDD should the BTC chaindata be on? I just randomly create a folder?
(This is because when I read the whole section on Bitcoin tab in the guide, it teaches how to make the directory as if we were downloading and preparing the node all from the PI itself. Correct me not?)

3. After the node has been transferred across, is it only then we continue with the instructions from the guide? 

In summary,my question is , in which phase of the guide do i transfer the BTC from local to the formatted HDD , wherein the HDD to transfer to and wherein the guide to continue from after transfer successful?

I may missed out completely but i've been reading the document , troubleshooting pages and all issues (open & closed) .

Thanks advance.

anonymoustrident, 2019-03-26:
Anybody knows when and how should we transfer across the fully synced BTC node to the PI?

openoms, 2019-03-26:
Hi! If your HDD is already formatted with the Pi you can plug your HDD into your Mac and copy the blockchaindata onto it.
Look for the folder on your Mac containing the folders: blocks, chainstate and indexes.
Copy these three folders to your HDD in  a /bitcoin folder. That is the root of your HDD /bitcoin.
Once plugged back in to the Pi, log in through ssh and make sure the folder is owned by the bitcoin user by running:
`sudo chown -R bitcoin:bitcoin /mnt/hdd/bitcoin`

anonymoustrident, 2019-03-26:
Thank you @openoms for your reply. I get it .so after I plugged back the fully synced node into the pi, do I still need to run through the “bitcoin node” steps mentioned in the guide ? Or I can go straight to LND?

openoms, 2019-03-26:
You need to do everything in the guide. This would only take care of the  `Copy the Mainnet Blockchain` part:  https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#copy-the-mainnet-blockchain

anonymoustrident, 2019-03-26:
> Hi! If your HDD is already formatted with the Pi you can plug your HDD into your Mac and copy the blockchaindata onto it.
> Look for the folder on your Mac containing the folders: blocks, chainstate [...]

@openoms ,Just fyi, the HDD is no longer recognise by the mac due to the format from the PI. EXT4?I think the next best way is to SSH those 3 files into the pi ?

anonymoustrident, 2019-03-27:
> You need to do everything in the guide. This would only take care of the `Copy the Mainnet Blockchain` part: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#copy-the-mainnet-blockchain

AH !,  that explains. Thanks a ton. So previously the guide was running through testnet ? Okay, if that is the case, I guess, I am not just ready yet to transfer into the pi. Will finish the bitcoin steps first.

anonymoustrident, 2019-03-27:
Just another question before I close this 

> Among other infos, the “verificationprogress” is shown. Once this value reaches almost 1 (0.999…), the blockchain is up-to-date and fully validated.

So while waiting for it to be fully sync to the testnet at verificationprogress=1 , I will not be able to proceed with the LND setup , only able to do so after the chain is fully sync. Can confirm?

openoms, 2019-03-27:
> Just fyi, the HDD is no longer recognise by the mac due to the format from the PI. EXT4?I think the next best way is to SSH those 3 files into the pi ?

You could make the Mac able to deal with EXT4 with this software for example: https://github.com/osxfuse/osxfuse/wiki/Ext . Please note that this is not a tested method, so the copying with SCP would be the most secure option and would need no modification on your machine.

> So while waiting for it to be fully sync to the testnet at verificationprogress=1 , I will not be able to proceed with the LND setup , only able to do so after the chain is fully sync. Can confirm?

You can carry on installing Lightning, but you will not be able to test it until the blockchain is fully synced (either on test or on mainnet).
If you follow the guide the synchronisation is happening as a background process, so you can proceed further and come back later to have look at the log.

anonymoustrident, 2019-03-28:
> You could make the Mac able to deal with EXT4 with this software for example: https://github.com/osxfuse/osxfuse/wiki/Ext . Please note that this is not a tested method, so the copying with SCP [...]

Thanks a ton @openoms ! That really helps in the understanding from the already very detailed instructions.  Allow me to keep it open just yet, till i successfully sync the main net over. 

Right now, I just autostart LND and should be able to port over to mainnet this weekend after a few tries on the testnet.

Stadicus, 2019-04-18:
Thanks @openoms for having my back when I'm slow in responding. My apologies @anonymoustrident.

I think everything has been said. First, the Bitcoin testnet is synced directly on the Pi (that takes only a few hours) and only after you're comfortable with everything you proceed to copy the mainnet blockchain with ssh or WinSCP.

Did that work for you?

anonymoustrident, 2019-04-22:
Thanks @Stadicus for replying nevertheless. 

Yes , testnet worked well for me. Right now I am still figuring out on the mainnet stuff.

So this is what I did. I stop the bitcoin service, copy over the files "Blocks", "Chainstate" and "Indexes"  over from my HDD to the Pi HDD. 

This is suppose to be fully synced but when i check blockchaininfo , it shows 

```
"chain": "main",
  "blocks": 425742,
  "headers": 572686,
  "bestblockhash": "0000000000000000009c0df94f533001cf0f2ff9dd7804bd264c7299f6800783",
  "difficulty": 217375482757.2376,
  "mediantime": 1471528964,
  "verificationprogress": **0.3884505949533498**,
  "initialblockdownload": true,
  "chainwork": "000000000000000000000000000000000000000000236fc67e3e178a3d185ab7",
  "size_on_disk": 91789806802,
  "pruned": false,
  "softforks": [
    {
      "id": "bip34",
      "version": 2,
      "reject": {
        "status": true
      }
    },
    {
      "id": "bip66",
      "version": 3,
      "reject": {
        "status": true
      }
    },
    {
      "id": "bip65",
      "version": 4,
      "reject": {
        "status": true
      }
    }
  ],
  "bip9_softforks": {
    "csv": {
      "status": "active",
      "startTime": 1462060800,
      "timeout": 1493596800,
      "since": 419328
    },
    "segwit": {
      "status": "defined",
      "startTime": 1479168000,
      "timeout": 1510704000,
      "since": 0
    }
  },
  "warnings": ""
}

```

So I'm wondering if there is something wrong mid way? I am planning to delete the whole mainnet files and re-transfer later this week.

Just to check, I didnt create any folder at all for the main net and just paste the files as per how it was created in the default location on the local PC.

Because I realised there is a folder for testnet but not for Mainnet.

Stadicus, 2019-04-25:
That's strange as usually when something goes wrong, verification starts again at 0%, but you are already at 30%+ (that would take days). Also, there are many blocks present, but over 100k missing...

I would first check the download & status on your regular computer to make sure it's actually fully caught up, and then recopy the data.

sentinel12, 2019-04-27:
Yes, I guess I need to re-do everything. Got this today .

```
$ bitcoin-cli getblockchaininfo
error code: -28
error message:
Loading wallet...
```

```
2019-04-26T16:01:40Z Bitcoin Core version v0.17.1 (release build)
2019-04-26T16:01:40Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-04-26T16:01:40Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-04-26T16:01:40Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-04-26T16:01:40Z Using the 'standard' SHA256 implementation
2019-04-26T16:01:40Z Default data directory /home/bitcoin/.bitcoin
2019-04-26T16:01:40Z Using data directory /home/bitcoin/.bitcoin
2019-04-26T16:01:40Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-04-26T16:01:40Z Using at most 125 automatic connections (1024 file descriptors available)
2019-04-26T16:01:40Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-04-26T16:01:56Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-04-26T16:01:56Z Using 4 threads for script verification
2019-04-26T16:02:03Z scheduler thread start
2019-04-26T16:02:03Z Binding RPC on address :: port 8332 failed.
2019-04-26T16:02:04Z Binding RPC on address 0.0.0.0 port 8332 failed.
2019-04-26T16:02:04Z Unable to bind any endpoint for RPC server
2019-04-26T16:02:04Z Error: Unable to start HTTP server. See debug log for details.
2019-04-26T16:02:04Z Shutdown: In progress...
2019-04-26T16:02:04Z scheduler thread interrupt
2019-04-26T16:02:04Z Shutdown: done
```

Not sure why but what i did was to delete everything on the main chain folder and re transfer via ssh over.

Will try once more.

anonymoustrident, 2019-04-27:
So the whole main chain is corrupted. 

```
$ bitcoin-cli getblockchaininfo
error code: -28
error message:
Loading wallet...
```

```
2019-04-26T15:54:32Z Bitcoin Core version v0.17.1 (release build)
2019-04-26T15:54:32Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-04-26T15:54:32Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-04-26T15:54:32Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-04-26T15:54:32Z Using the 'standard' SHA256 implementation
2019-04-26T15:54:33Z Default data directory /home/bitcoin/.bitcoin
2019-04-26T15:54:33Z Using data directory /home/bitcoin/.bitcoin
2019-04-26T15:54:33Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-04-26T15:54:33Z Using at most 125 automatic connections (1024 file descriptors available)
2019-04-26T15:54:33Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-04-26T15:54:47Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-04-26T15:54:47Z Using 4 threads for script verification
2019-04-26T15:54:57Z scheduler thread start
2019-04-26T15:54:57Z Binding RPC on address :: port 8332 failed.
2019-04-26T15:54:57Z Binding RPC on address 0.0.0.0 port 8332 failed.
2019-04-26T15:54:57Z Unable to bind any endpoint for RPC server
2019-04-26T15:54:57Z Error: Unable to start HTTP server. See debug log for details.
2019-04-26T15:54:57Z Shutdown: In progress...
2019-04-26T15:54:57Z scheduler thread interrupt
2019-04-26T15:54:57Z Shutdown: done
```

Not sure for whatever reason. What i did was to delete everything in the main chain folder and re-copy it over via SSH. The copied files are fully synced on a local machine.

Will try it once more. 
Or maybe if it will be better to plug in the external HDD to the pi itself and copy over.

Any comments?

anonymoustrident, 2019-04-27:
> So the whole main chain is corrupted.
>
> ```
> $ bitcoin-cli getblockchaininfo
> error code: -28
> error message:
> Loading wallet...
> ```
> ```
> 2019-04-26T15:54:32Z Bitcoin Core version v0.17.1 (release build) [...]

Additional log 

```
2019-04-27T06:52:20Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-04-27T06:52:22Z Using 4 threads for script verification
2019-04-27T06:52:30Z scheduler thread start
2019-04-27T06:52:30Z Binding RPC on address :: port 8332 failed.
2019-04-27T06:52:30Z Binding RPC on address 0.0.0.0 port 8332 failed.
2019-04-27T06:52:30Z Unable to bind any endpoint for RPC server
2019-04-27T06:52:30Z Error: Unable to start HTTP server. See debug log for details.
2019-04-27T06:52:30Z Shutdown: In progress...
2019-04-27T06:52:30Z scheduler thread interrupt
2019-04-27T06:52:30Z Shutdown: done
```

Stadicus, 2019-04-27:
The error message looks more like bitcoind not being able to claim port 8332 for itself (binding to it).

Did you restart the Pi already? Then you can check ports already in use like this:
`sudo lsof -i -P -n | grep LISTEN`

anonymoustrident, 2019-04-27:
Thanks @Stadicus , sudo lsof -i -P -n | grep LISTEN this command help me to solve it. I did restart and try to update some nodes from the synced file. Thanks , it is now 0.99 verification and soon to catch up to the latest height.

closed 2019-04-27 by anonymoustrident