raspibolt/raspibolt #420: The guide is missing some ports under the firewall section

Phillo, 2019-03-31:
Hi Stadicus,

First off, this is one HELL of a good guide, and I am willing to let some sats flow your way for your hard work... BUT only once my problem is solved.

I have been getting the famous error: Unable to discover a NAT-PMP enabled device on the local network.

It took me a while to realise that I had omitted one of the ufw firewall rules, namely:

$ ufw allow proto udp from 192.168.0.0/24 port 1900 to any comment 'allow local LAN SSDP for UPnP discovery'

So this led to the first little bug: This rule is not visible in the screenshot UNDERNEATH the list of rules. The image can be found at: https://raw.githubusercontent.com/Stadicus/guides/master/raspibolt/images/20_ufw_status.png.

I went on to correct that mistake. In so doing I turned off the firewall whilst testing, and lo and behold: UPnP is all good (on my second router, my fibre optic router actually has a problem with UPnP which was not helping.

Happy I had found the source, I went and enabled ufw again... and... failure!! I think you have forgotten one or more other ports that need to be opened. Could it be that port 10009 ALSO has to be given access on the local LAN?

I will continue investigating, but it beggars the question how ANYBODY else got UPnP to work if they implemented the firewall rules to the letter like in your great guide...

Anyway, thought I would let you know... I will come back here if I work it out on my own... I don't really want to be on mainnet with a disabled firewall...

Thanks,
Phil

Phillo, 2019-03-31:
I have kept up with the investigation. When the firewall is on I get the following:

lnd[2159]: 2019-03-31 19:29:05.265 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: read udp 192.168.100.95:48015->192.168.100.1:5351: recvfrom: connection refused
systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE

My current firewall settings are:

To                         Action      From

22                         ALLOW       192.168.100.0/24           # allow SSH from local LAN
Anywhere                   ALLOW       192.168.100.0/24 1900/udp  # allow local LAN SSDP for UPnP discovery
9735                       ALLOW       Anywhere                   # allow Lightning
8333                       ALLOW       Anywhere                   # allow Bitcoin mainnet
18333                      ALLOW       Anywhere                   # allow Bitcoin testnet
9736                       ALLOW       Anywhere                   # allow Lightning
Anywhere                   ALLOW       192.168.0.0/24 1900/udp    # allow local LAN SSDP for UPnP discovery
Anywhere                   ALLOW       192.168.0.0/24 10009       # allow local LAN RCP connections
5351                       ALLOW       192.168.0.0/24             # allow NAT-PMP connections
9735 (v6)                  ALLOW       Anywhere (v6)              # allow Lightning
8333 (v6)                  ALLOW       Anywhere (v6)              # allow Bitcoin mainnet
18333 (v6)                 ALLOW       Anywhere (v6)              # allow Bitcoin testnet
9736 (v6)                  ALLOW       Anywhere (v6)              # allow Lightning

These rules are not open enough as the failure only shows up when ufw is enabled... Anyone has an idea what ports / IPs need to be opened? I made up the port 5351 rule myself thinking it was needed

giulnz, 2019-04-10:
Yes the 20_ufw_status.png is not update.
But raspibolt guide as it is now is correct. No need to extra ufw rules.
Check if you set in /home/bitcoin/.lnd/lnd.conf   nat=true
If you still have problem your router probably doesn't support UPnP, if so delete that line (nat=true)
If lnd.service still not start try this https://github.com/robclark56/RaspiBolt-Extras/blob/master/RB_extra_02.md

giulnz, 2019-04-10:
If you have static ip, you don't need getpublicipscript, just add this line 
--externalip=YOUR_IP
to /etc/systemd/system/lnd.service under [Service]
stop and restart lnd.service

Lnd.service will not start without an ip, so that why you have lnd.service: Main process exited, code=exited, status=1/FAILURE

Stadicus, 2019-04-18:
Hej @Phillo, did that solve your issue? UPnP is no silver bullet, there can be circumstances where network config or funky routers make configuration really hard...

closed 2019-04-25 by Stadicus