raspibolt/raspibolt #423: lncli unlock no longer authenticating

beeforbacon1, 2019-04-10:
- admin@berrybolt:~ $ lncli unlock
Input wallet password:
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority (possibly because of \"x509: ECDSA verification failure\" while trying to verify candidate authority certificate \"berrybolt\")"

--------

this just randomly started happening, after flawlessly running raspibolt from this guide since january. what could be a fix ? Or what further information should I provide? Thanks.

daniel-vera-g, 2019-04-14:
1. What lnd version are you running?

It could be, that you generated new lnd authentication tokens(macaroons) and those are not in the admin folder, only in the bitcond folder.

Try Coping permission files and TLS cert to user "admin"

```
$2 cd /home/bitcoin/
$2 sudo cp --parents .lnd/data/chain/bitcoin/testnet/admin.macaroon /home/admin/
$2 sudo cp /home/bitcoin/.lnd/tls.cert /home/admin/.lnd
$2 sudo chown -R admin:admin /home/admin/.lnd/
```

Ldngzero, 2019-05-25:
I've got similar issue:

_admin ~ ฿ lncli unlock
Input wallet password:
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate has expired or is not yet valid"_

I am running lnd 0.6.1, bitcoind 18.0. 
Permission files and TLS cert copied to user "admin".
Where to look for a solution? Thanks.

nipponese, 2020-01-01:
FYI, It sounds like OP is not using testnet.

closed 2020-02-28 by Stadicus