raspibolt/raspibolt #427: Better explanation - Copy bitcoin.conf to user "admin" home

Janaka-Steph, 2019-04-13:
Hello,

Thundroid guide recommend to copy bitcoin.conf to user "admin" home directory without explaining much the purpose of it. It would be nice to have a bit more explanation. 
Thank you
 
```
Copy bitcoin.conf to user "admin" home directory for RPC credentials
$ mkdir /home/admin/.bitcoin
$ sudo cp /home/bitcoin/.bitcoin/bitcoin.conf /home/admin/.bitcoin/
$ sudo chown admin:admin /home/admin/.bitcoin/bitcoin.conf
```

daniel-vera-g, 2019-04-14:
Well I think `Copy bitcoin.conf to user "admin" home directory for RPC credentials` is enough explanation.

You do this after creating the service to autostart bitcoind. Since the bitcoin.conf belongs to the user bitcoin, but the user admin autostarts the bitcoind.service, the config file has to be copied over to the admin user.

If you still think it's not enough explanation, you can make a proposal of what should be added and I can make a PR for you.

Janaka-Steph, 2019-04-15:
Hello, 
Thank you for your quick reply, I understand better know. I think what you just said would be a good addition "Since the bitcoin.conf belongs to the user bitcoin, but the user admin autostarts the bitcoind.service, the config file has to be copied over to the admin user."   
It's just a suggestion, I guess it's not clear for everybody but feel free to close the issue.

daniel-vera-g, 2019-04-15:
I think it's enough. This issue can be a reference for people which don't understand it. Will leave @Stadicus the decision.

Stadicus, 2019-04-18:
I have to admin that I am currently not working actively on the Thundroid guide. The RaspiBolt is basically the same, so you can always cross check there.

closed 2019-04-18 by Stadicus