raspibolt/raspibolt #430: Symbolic link missing

kev-vl, 2019-04-18:
Hi,

After experiencing some trouble with LND I decided to start over with LND. However, I'm running into problems when trying to setup a symbolic link:

`bitcoin@FatTonyNode:~ $ ls -la`
`drwxr-xr-x 2 bitcoin bitcoin 4096 Apr 18 16:00 .lnd`

This presents .lnd in dark blue, as expected. Note: Symbolic links for .bitcoin and eps are actually showing.

Running this command does nothing to create a symbolic link:

`bitcoin@FatTonyNode:~ $ ln -s /home/bitcoin/.lnd /mnt/hdd/lnd`
`drwxr-xr-x 2 bitcoin bitcoin 4096 Apr 18 16:00 .lnd`

What could this be? I'm thinking of getting LND out of there altogether so I can start over from a clean slate. How would I go about that properly?

daniel-vera-g, 2019-04-18:
What do you get, when you try to slimlink the directory?
Does the directory exist on the hard drive?
What do you get when you do: 'ls - a /home/bitcoin/.lnd/'

kev-vl, 2019-04-18:
Running `bitcoin@FatTonyNode:~ $ ln -s /home/bitcoin/.lnd /mnt/hdd/lnd` returns `ln: failed to create symbolic link '/mnt/hdd/lnd/.lnd': File exists` but then there is no reference of this link after running `ls -la`.

Full result of `ls -la`:
```
total 52
drwxr-xr-x 7 bitcoin bitcoin 4096 Apr 18 16:18 .
drwxr-xr-x 5 root    root    4096 Mar 30 11:52 ..
-rw------- 1 bitcoin bitcoin 4574 Apr 18 16:03 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Mar 30 11:52 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Mar 30 11:52 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   16 Mar 30 14:36 .bitcoin -> /mnt/hdd/bitcoin
drwxr-xr-x 3 bitcoin bitcoin 4096 Mar 30 22:15 .cache
lrwxrwxrwx 1 bitcoin bitcoin   33 Mar 30 22:07 electrum-personal-server -> /mnt/hdd/electrum-personal-server
drwx------ 3 bitcoin bitcoin 4096 Mar 30 22:10 .gnupg
drwxr-xr-x 2 bitcoin bitcoin 4096 Apr 18 16:00 .lnd
drwx------ 6 bitcoin bitcoin 4096 Mar 30 22:15 .local
drwxr-xr-x 2 bitcoin bitcoin 4096 Mar 30 14:36 .nano
-rw-r--r-- 1 bitcoin bitcoin  675 Mar 30 11:52 .profile
-rw-r--r-- 1 bitcoin bitcoin    0 Apr 18 16:18 test
-rw-r--r-- 1 bitcoin bitcoin  259 Mar 30 22:09 .wget-hsts
```
Re: your question
`bitcoin@FatTonyNode:~ $ ls -a /home/bitcoin/.lnd/`
`.  ..  lnd`

Thanks

daniel-vera-g, 2019-04-18:
Well, try deleting `.lnd`:

`rm /home/bitcoin/.lnd` or `rm -rf /home/bitcoin/.lnd`

And try again simlinking ;)

Maybe you did something wrong at first, created the simlink and then could not create the simlink another time since it's already there.

kev-vl, 2019-04-18:
It must have been something along those lines, as completely deleting /home/bitcoin/.lnd did the trick. Got my LND-node up and running again, over Tor this time!

Thanks for your help :), closing this issue

closed 2019-04-18 by kev-vl