raspibolt/raspibolt #431: mkdir /mnt/hdd/electrum-personal-server

walletofsatoshi, 2019-04-18:


Stadicus, 2019-04-18:
All `sudo` commands need to be run as user "admin". The other users are resticted from sudo permissions for security reasons.

walletofsatoshi, 2019-04-18:
It’s a raspiblitz setup had to edit sudoers to allow bitcoin as user to get it working. But having now issues running the server getting error as config file non existent despite following and editing saving the simple file do we have to specify datadir in the config file?

Stadicus, 2019-04-18:
I am not familiar with the RaspiBlitz in detail, I suggest posting the question directly to their repository. 

In general it's safe to assume that the "bitcoin" user is not supposed to be sudo. Isn't there a "admin" or "root" user for that purpose?

walletofsatoshi, 2019-04-18:
I used your guide to get EPS Installed which has user as bitcoin there is no separate guide fir raspi blitz EPS yet . I finally seem to get it working will keep you posted

closed 2019-04-24 by walletofsatoshi