raspibolt/raspibolt #432: Update to lnd v0.6-beta

Cees2daBees, 2019-04-19:
When upgrading LND in the past using the guide here https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md, I would follow the same steps and just modify the text to the latest version.

Can I do the same with the 0.6 update? Reading the release notes (https://github.com/lightningnetwork/lnd/releases) they make mention of things that are different in 0.6 and I'm not certain I could just follow the same steps in this guide? Don't want to royally screw things up so thought I'd ask first.

Thanks!

daniel-vera-g, 2019-04-19:
Yes, it's possible without problems.  
I've updated and it worked everything perfectly. 
They are some steps, that are not needed because they refer to previous versions. 
I've made a gist with the commands to update: 

https://gist.github.com/daniel-vera-g/92b33ffc6b7d4754056254f92ede4a1b

Feel free to check out ;-)

Cees2daBees, 2019-04-19:
@daniel-vera-g Thank you kind sir!! Upgrade complete!!

closed 2019-04-19 by Cees2daBees