raspibolt/raspibolt #433: Web GUI Bitcoin Full Node

OpenSystemCode, 2019-04-19:
Hey Guys,

First of all  @Stadicus great guide, you rock buddy.

I want install some Web GUI for my Bitcoin Full Node and found this Project:

https://github.com/KanoczTomas/01Cnode

I followed the instructions. I stuck at npm install cmd.

Log output:

```
31673 warn optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
31674 warn notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
31675 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Valid OS:    darwin
31675 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Valid Arch:  any
31675 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Actual OS:   linux
31675 verbose notsup SKIPPING OPTIONAL DEPENDENCY: Actual Arch: arm
31676 verbose stack Error: zeromq@4.6.0 install: `node scripts/prebuild-install.js || (node scripts/preinstall.js && node-gyp rebuild)`
31676 verbose stack Exit status 1
31676 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
31676 verbose stack     at EventEmitter.emit (events.js:193:13)
31676 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
31676 verbose stack     at ChildProcess.emit (events.js:193:13)
31676 verbose stack     at maybeClose (internal/child_process.js:999:16)
31676 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:266:5)
31677 verbose pkgid zeromq@4.6.0
31678 verbose cwd /home/admin/bitcoin-dashboard
31679 verbose Linux 4.14.70-v7+
31680 verbose argv "/usr/bin/node" "/usr/bin/npm" "install"
31681 verbose node v11.14.0
31682 verbose npm  v6.7.0
31683 error code ELIFECYCLE
31684 error errno 1
31685 error zeromq@4.6.0 install: `node scripts/prebuild-install.js || (node scripts/preinstall.js && node-gyp rebuild)`
31685 error Exit status 1
31686 error Failed at the zeromq@4.6.0 install script.
31686 error This is probably not a problem with npm. There is likely additional logging output above.
31687 verbose exit [ 1, true ]
```

Nodejs Stuff:

admin@BitcoinJesus:~ $ node -v
11.14.0

admin@BitcoinJesus:~ $ npm -v
v6.7.0

Can someone help me or point me in the right direction?

Best regards.

daniel-vera-g, 2019-04-19:
Hey, I'm sorry to disappoint you but as I can see this is not an issue with the raspibolt.

It's better to post your issue directly on the repository of the web GUI. They can probably help you more ;-)

Stadicus, 2019-04-25:
I am absolutely not an expert with Node.js, sorry. Please ask directly in the project's repository.

closed 2019-04-25 by Stadicus