raspibolt/raspibolt #437: ufw settings causing connection time-out

Drewpower, 2019-04-24:
When writing the uncomplicated firewall I lose the ability to connect ssh.  If I reconnect using a display and disable fsw I am able to reconnect ssh.  I've played around with configurations but have not found the fix.  
[image]
[image]

I'm using dhcp allocation over ethernet LAN and a static iPv4 address.

Stadicus, 2019-04-25:
You are probably blocking port 22. What is the IP address of your Pi and your regular computer?

Drewpower, 2019-04-25:
Okay....
The Pi IP is 192.168.1.120 and
My regular computer is 192.168.1.103

daniel-vera-g, 2019-04-26:
> Okay.... The Pi IP is 192.168.1.120 and My regular computer is 192.168.1.103
> […](#)

What is your IP-Range?

My Router has: `192.168.2.0\24`. That means he can give addresses between `192.168.2.0` and `192.168.2.255`. Therefore I've `192.168.2.0` in the ufw settings.

As I can see from the Image you've got the IP-Address range of: `192.168.1.0\24`. So you have to set `192.168.1.0\24` in the first two rows of the `from` column.

Drewpower, 2019-04-29:
Solved:
I went in and did ufw reset
Input using 192.168.1.0\24 in first row and now working.
thank you

daniel-vera-g, 2019-04-29:
> Solved:
> I went in and did ufw reset
> Input using 192.168.1.0\24 in first row and now working.
> thank you

Perfect, no problem ;-)

@Stadicus #437 can be closed

closed 2019-04-29 by Stadicus