raspibolt/raspibolt #440: Connect the Zeus Wallet Android to the RaspiBolt over Tor

janglorumo, 2019-04-27:
Hello! I saw this guide, and wanted to be able to use my Android + Tor to spend from my Lightning Node. I am not able to get very far and after some hours of attempting to troubleshoot in the Zeus Wallet telegram, I have come here.

https://github.com/openoms/bitcoin-tutorials/blob/master/Zeus_to_RaspiBlitz_through_Tor.md

```
Edit torrc with sudo nano /etc/tor/torrc and add the following lines (myandroid can be unique):

HiddenServiceDir /mnt/hdd/tor/lnd_api/
HiddenServiceVersion 2
HiddenServiceAuthorizeClient stealth myandroid
HiddenServicePort 8080 127.0.0.1:8080
HiddenServicePort 10009 127.0.0.1:10009
```

After inserting this into torrc and preforming systemctl restart tor my bitcoind goes into failure. Logs show this for tor + bitcoind 

```
Apr 26 20:15:04 username systemd[1]: Started Anonymizing overlay network for TCP (multi-instance-master).
2019-04-25T19:15:13Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
```

pardus79, 2019-06-11:
Try this instead:
```
HiddenServiceDir /var/lib/tor/lnd_api/
HiddenServiceVersion 2
HiddenServiceAuthorizeClient stealth myandroid
HiddenServicePort 8080 127.0.0.1:8080  
HiddenServicePort 10009 127.0.0.1:10009
```

pardus79, 2019-06-11:
Try again using #454 as a guide.

closed 2020-02-28 by Stadicus