raspibolt/raspibolt #441: LND Mainnet Issues

anonymoustrident, 2019-04-27:
Hi everyone, 

I am finishing the guide on mainnet but LND issue arise when I am on this section.

```
Copy permission files and TLS cert to user “admin” to use lncli

$ sudo cp /home/bitcoin/.lnd/tls.cert /home/admin/.lnd  
$ cd /home/bitcoin/  
$ sudo cp --parents .lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/admin/
$ sudo chown admin:admin /home/admin/.lnd/ -R  
```

The file does not exist for `sudo cp --parents .lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/admin/`. I have used putty to search and there is indeed no file at all in /home/bitcoin

Everything worked well on testnet. But on main net, i am facing this issue. Should I rm -rf /home/bitcoin/.lnd/ delete everything ?

daniel-vera-g, 2019-04-28:
Are you sure, that you started bitcoind and lnd in mainnet mode and **created a new mainnet wallet** on lnd ?

I think, that the macaroon files get created when you create a lnd wallet. Maybe you forgot this step.

If you don't have funds on the lnd node, you could try everything again and delete  the `. lnd` folder. But first check, that you didn't miss any step.

anonymoustrident, 2019-04-28:
Are you sure, that you started bitcoind (Yes, I am sure)
Are you sure, lnd in mainnet mode (yes, I am sure)
created a new mainnet wallet on lnd ? _(How can I check that the creation of the new wallet was on Mainnet and not testnet?)_

Perharps when i do cli-create, it is creating a testnet wallet? Let me show.

anonymoustrident, 2019-04-28:
> Are you sure, that you started bitcoind (Yes, I am sure)
> Are you sure, lnd in mainnet mode (yes, I am sure)
> created a new mainnet wallet on lnd ? _(How can I check that the creation of the new [...]
```
bitcoin@raspberrypi:~ $ lncli create
Input wallet password: 
Confirm wallet password: 

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase): 

Generating fresh cipher seed...

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!

---------------BEGIN LND CIPHER SEED---------------
 1. able      2. airport   3. sheriff   4. apology
 5. season    6. depend    7. menu      8. crouch 
 9. address  10. you      11. trash    12. conduct   
21. merit    22. inch     23. ranch    24. bring  
---------------END LND CIPHER SEED-----------------

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!
[lncli] rpc error: code = Unknown desc = password must have at least 8 characters

```

Here, the creation of the new wallet. How can i check which net is this on?

anonymoustrident, 2019-04-28:
@daniel-vera-g 

This is the blockchain info 
```
bitcoin@raspberrypi:~ $ bitcoin-cli getblockchaininfo
{
  "chain": "main",
  "blocks": 573565,
  "headers": 573565,
  "bestblockhash": "000000000000000000268076fdba254a48e9315648ce651d43f254910fd02675",
  "difficulty": 6353030562983.983,
  "mediantime": 1556426326,
  "verificationprogress": 0.9999978310136242,
  "initialblockdownload": false,
  "chainwork": "000000000000000000000000000000000000000005feff8d4c656140ab537796",

```

This is the LND info

```
bitcoin@raspberrypi:~ $ lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory
```

You are right. Seems like i need to redo LND again? Because there is some issues with LND main net. Is that why I am always on LND Test net?

daniel-vera-g, 2019-04-28:
Check the last line of your `lncli create` output:

> [lncli] rpc error: code = Unknown desc = password must have at least 8 characters

Check your password for having at least 8 characters and try again.

After successfully creating the wallet without errors, check with `lncli getinfo` and to check if you're on mainnet or testnet: `lncli getinfo | grep testnet`

Stadicus, 2019-04-29:
That hidden error rings a bell, had the same issue myself... 
https://github.com/lightningnetwork/lnd/issues/890#issuecomment-374762415

anonymoustrident, 2019-04-30:
Just to check some stuff if these are normal 

```
sudo ls -la /home/bitcoin/.lnd/data/chain/bitcoin/testnet/total 154392
drwx------ 2 bitcoin bitcoin      4096 Mar 27 23:32 .
drwx------ 4 bitcoin bitcoin      4096 Apr 30 21:49 ..
-rw------- 1 bitcoin bitcoin       232 Mar 27 23:32 admin.macaroon
-rw-r--r-- 1 bitcoin bitcoin       115 Mar 27 23:32 invoice.macaroon
-rw------- 1 bitcoin bitcoin     32768 Apr 13 02:12 macaroons.db
-rw-r--r-- 1 bitcoin bitcoin       183 Mar 27 23:32 readonly.macaroon
-rw------- 1 bitcoin bitcoin 174792704 Apr 13 02:27 wallet.db
```
Macaroon found on test
```
sudo ls -la /home/bitcoin/.lnd/
total 28
drwxr-xr-x 4 bitcoin bitcoin 4096 Apr 30 21:47 .
drwxr-xr-x 5 bitcoin bitcoin 4096 Mar 27 23:14 ..
drwx------ 4 bitcoin bitcoin 4096 Mar 27 23:32 data
lrwxrwxrwx 1 bitcoin bitcoin   12 Apr 30 21:46 lnd -> /mnt/hdd/lnd
-rw-r--r-- 1 bitcoin bitcoin  515 Apr 17 07:37 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Mar 27 23:29 logs
-rw-r--r-- 1 bitcoin bitcoin    0 Apr 14 00:44 test
-rw-r--r-- 1 bitcoin bitcoin  749 Apr 13 01:04 tls.cert
-rw------- 1 bitcoin bitcoin  227 Apr 13 01:04 tls.key
```
Macaroon found on main
```
sudo ls -la /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/total 6332
drwx------ 2 bitcoin bitcoin    4096 Apr 30 21:49 .
drwx------ 4 bitcoin bitcoin    4096 Apr 30 21:49 ..
-rw------- 1 bitcoin bitcoin     232 Apr 30 21:49 admin.macaroon
-rw-r--r-- 1 bitcoin bitcoin     115 Apr 30 21:49 invoice.macaroon
-rw------- 1 bitcoin bitcoin   32768 Apr 30 21:49 macaroons.db
-rw-r--r-- 1 bitcoin bitcoin     183 Apr 30 21:49 readonly.macaroon
-rw------- 1 bitcoin bitcoin 8388608 Apr 30 21:55 wallet.db
```
LND Info 
```
lncli --network=mainnet getinfo
{
    "identity_pubkey": "03e10e844935c53f70ba2b9f276a70700445a",
    "alias": "XXXXX",
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_peers": 0,
    "block_height": 573933,
    "block_hash": "00000000000000000017d53334d5a079bdcd179",
    "synced_to_chain": false,
    "testnet": false,
    "chains": [
        "bitcoin"
    ],
    "uris": [
        "03e10e844935c53f70859b29f276a70700445a@27.104.173.224:9735"
    ],
    "best_header_timestamp": "1292453136",
    "version": "0.5.2-beta commit=v0.5.2-beta",
    "num_inactive_channels": 0
```

Thanks everyone for the replies.

anonymoustrident, 2019-05-01:
This morning woke up to this 

```
May 01 09:46:25 raspberrypi lnd[2176]: 2019-05-01 09:46:25.088 [INF] LNWL: Caught up to height 570000
May 01 09:46:42 raspberrypi lnd[2176]: 2019-05-01 09:46:42.827 [INF] LNWL: Started rescan from block 0000000000000000000a0f1515a202c99a131126a42d8d5356eb92f27b3642f2 (height 573505) for 0 addresses
May 01 09:46:42 raspberrypi lnd[2176]: 2019-05-01 09:46:42.827 [INF] LNWL: Starting rescan from block 0000000000000000000a0f1515a202c99a131126a42d8d5356eb92f27b3642f2
May 01 09:52:22 raspberrypi lnd[2176]: 2019-05-01 09:52:22.550 [ERR] LNWL: Failed to process consensus server notification (name: `blockconnected`, detail: `failed to store sync information 00000000000000000000ca3c873083c4ddd1be4501324eca5022bfbe315cc12a: failed to fetch block hash for height 574013: block not found`)
**May 01 09:53:30 raspberrypi lnd[2176]: 2019-05-01 09:53:30.321 [ERR] LNWL: Failed to process consensus server notification (name: `blockconnected`, detail: `failed to store sync information 00000000000000000005581dd5312b7272011313ebad9a38763a6f88ac9b4f30: failed to fetch block hash for height 574014: block not found`)**
```

Somehow it cant sync up the final bits.

closed 2020-02-28 by Stadicus