raspibolt/raspibolt #443: LND TOR cannot receive payments and cannot open channels to my lnd

kriloter, 2019-05-03:
Hello guys, I have working bitcoind lnd lnd hub tor based on stadicius guide, I can pay invoices from LN wallet but I can't receive payments and open incomming channels. I have only 3 outbound channels open ... Port forwarding and fw is ok I think, any suggestions to troubleshoot where is the problem? Thank you very much ... I spent few days and nothing ...
My node is krilnd, 25wvn35yaxdosrfr2xbkg3xpmfsbzj2ufrc7n5zy6dhq5eukfy2xqiyd.onion:9735 . If you be so kind and try to connect ... Thanks

daniel-vera-g, 2019-05-03:
> Hello guys, I have working bitcoind lnd lnd hub tor based on stadicius guide, I can pay invoices from LN wallet but I can't receive payments and open incomming channels. I have only 3 outbound [...]

Some clarifying informations to begin:
 
1. "Clearnet"(With IP-address) nodes can't open connect channels to you.
2. You can open channels to Clearnet and Onion(Tor) nodes
3. Only onion(Tor) nodes can open channels to you
4. You do not need port forwarding using tor
5. It's rare, that people just open channels to you randomly.

Your node is perfectly reachable, I've already opened a connection to you.
When specifying your node, you should give your full URI = PUBKEY@ADRESS
In your case: `0205c7027fae92c447ba69fa512ac2cfc6a95d46a038f276f587d629dde42c8b5c@25wvn35yaxdosrfr2xbkg3xpmfsbzj2ufrc7n5zy6dhq5eukfy2xqiyd.onion:9735`(From: https://1ml.com/node/0205c7027fae92c447ba69fa512ac2cfc6a95d46a038f276f587d629dde42c8b5c) 

I think there's no issue here and @Stadicus can close it.

kriloter, 2019-05-03:
thank you very much.
it's question out of here but ... why I can't pay any invoice? why I can't pay invoce from blue wallet? It says [lncli] unable to find a path to destination
1. clear
2. clear
3. ok ...
4. understood

daniel-vera-g, 2019-05-04:
> thank you very much.
> it's question out of here but ... why I can't receive any payment? why I can't pay invoce from blue wallet? It says [lncli] unable to find a path to destination
> 
>     1. clear
> 
>     2. clear
> 
>     3. ok ...
> 
>     4. understood

This is probably an issue with blue wallet. You need some incoming Liquidity to be paid. If you pay your invoice, but your blue wallet node does not have inbound channels, the transaction cannot be made.

closed 2020-02-28 by Stadicus