raspibolt/raspibolt #453: can't issue sudo commands after the command: sudo chown -R bitcoin:bitcoin /mnt/hdd/

thermalCat, 2019-06-09:
I got as far as this step in mounting external disk:

$ sudo chown -R bitcoin:bitcoin /mnt/hdd/

after that, all sudo commands return the error:
 /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

Is it possible that there is an extra forward slash at the end of that command?
In my near-total ignorance of the linux system, I googled similar commands and they look more like:

$ sudo chown -R bitcoin:bitcoin /mnt/hdd

daniel-vera-g, 2019-06-10:
Hi, your Problem does not seem related to some issue in the tutorial. There is probably an issue with the permissions in your file system.

Try the answers here: https://askubuntu.com/questions/452860/usr-bin-sudo-must-be-owned-by-uid-0-and-have-the-setuid-bit-set

thermalCat, 2019-06-16:
thanks,
I had rather stupidly confused the mount directory of the SD card with that of the external drive.

closed 2019-06-16 by thermalCat