raspibolt/raspibolt #455: raspberry pi could not communicate with wpa_supplicant

Lockejoe, 2019-06-14:
Hello all,

After setting up my RaspiBlitz completely, I'm currently trying to change from ethernet connection to WI-FI.
However I can't get it to work.
First, I tried adding the wpa_supplicant.conf file manually with card reader with an other system, After that I tried via ssh and ethernet. I tried many different wpa_supplicant.conf versions, but if I'm opening raspi-config menu, I'm always getting "could not communicate with wpa_supplicant. There was an error running option N2 wi_fi" in Network Options Menu.

Hopefully, Someone can help.

Many thanks in advance.

daniel-vera-g, 2019-06-15:
> Hello all,
>
> After setting up my RaspiBlitz completely, I'm currently trying to change from ethernet connection to WI-FI.
> However I can't get it to work.
> First, I tried adding the wpa_supplicant.conf [...]

I suppose, you followed the raspibolt wifi tutorial part.

Make sure your country code is right. It often produces errors.

If not, try to update to the latest version & reboot:

1. Connect to wired option temporarily
2. Open terminal and type: sudo raspi-config
3. Select option 8 to update the pi
4. Reboot
5. You should now be able to right click and turn on wifi and remove the wired network connection

Lockejoe, 2019-06-15:
Hello,

Thanks for your answer.
Yes I followed the wifi tutorial part and I also did updates without any result.

I updated via raspi-config
then tried sudo apt-get update
then tried sudo apt-get dist-upgrade

What do you mean by "You should now be able to right click and turn on wifi and remove the wired network connection" ?

Any other ideas?

Stadicus, 2020-02-28:
Closing as "too old" :-) Feel free to reopen if still relevant.

closed 2020-02-28 by Stadicus

larryllix, 2020-06-11:
Deflecting this thread all over the place does not help the problem. All of those steps were followed but I still see this problem many places online. The setup process does not fix this problem.

My problem is exactly the same as the OP.  I have tried enabling permissions on the wpa_supplicant directories and .conf file to owner, group, and other = "rw" but raspi-config still can't access the wpa_supplicant information.

It should be noted my WiFi worked fine since inception of this distro, as well as upgrades for several months. It seems the last  _sudo apt-get update && sudo apt-get upgrade_  a week ago, broke this WiFi setup.

Pinktrap, 2020-06-13:
I have exactly the same issue.  If I connect a USB WiFi dongle the pi will connect but with the internal WiFi alone it will not connect since upgrading from Stretch to Buster.  wpa_supplicant contains contry code but raspi-config cannot see the wpa_supplicant file and hence (presumably) cannot pick up the code.

larryllix, 2020-06-13:
Thank you for your response.

I thought that also, that the upgrade of the O/S had a bug in it but I have taken the SD card and inserted it into another RPi 3B and the WiFi works just fine.

The other thought I had was I installed a robotic PiCar Hat and my WiFi failure seemed to star right about then.  I have since removed the Hat, tried everything I can find online with files and definitions.  I even low level reformatted the SD card and started over with a new Raspbian image, getting the same raspi-config failure every time.

I am thinking about trying the Hat on my other RPi 3B but afraid it will destroy that unit's WiFi also. I don't see how whatever I did with the GPIO pins could affect the WiFi section of the PCB though.

That leaves me with with a hardware problem not making sense and a software problem not making any sense either?????

Maybe I need to reinstall Raspbian from scratch again and not upgrade it to the latest version,this time. This would prove bugs in the latest upgrade or not. I am afraid this will be another RPi in the scrap. They are getting too expensive.

Pinktrap, 2020-06-16:
I eventually reloaded the Buster version from scratch (having previously done the upgrade from Stretch).  Lo and behold the WiFi worked immediately as before.  It must be a confllicting piece of software or setting but I couldn't track it down.  I hope you have similar good fortune, larrylix!

larryllix, 2020-06-16:
I switched my SD cards around with another RPi 3 and WiFi functioned again with that SD card. Figuring that that RPi wasn't updated in a while I nervously switched the two RPi functions. Then the WiFi quit on the other usage with the updted SD card. I figured it was the update that causes sudo raspi-config to break. I think I have confirmed that.
I also tried hooking up a monitor and KB and allowed the GUI to set up the WiFi, to no avail.

Then I re-installed Buster from scratch and found  WiFi working on one RPi again. I did not upgrade it again.

In the end I switched the two RPi 3B functions. One has a burned out WiFi section but is only used for my gGenealogy server software on wired Ethernet. 
The other has WiFi functioning and I nervously installed the same PiCar Hat on it with success.

I conclude that somehow I burned out the WiFi section of one RPI 3B abd I found another usage for that board.
I also conclude that the latest upgrade to Buster break raspi-config and it will not connect to wpa_supplicant. Yes I have tried every patch and test I could find online without success. and my country code is set, and the wpa_supplicant.conf file has been proofread and reedited from cut'n paste images a few dozen times...both of versions of them.

Yeah, RPI linux o/s is a disgusting mess with commands and options being pulled out of people's sleeves everywhere. Python3 is a matching disaster. Love them both. Hate them both.

erikoui, 2020-06-26:
Hello, I had the same problem. Turns out some software packages, in my case OpenMediaVault, interfere with raspbian's networking services.
I fixed the problem by using the openmediavault interface (under Network->Interfaces->add)
Hope this helps someone.

larryllix, 2020-06-26:
> 
> 
> Hello, I had the same problem. Turns out some software packages, in my case OpenMediaVault, interfere with raspbian's networking services.
> I fixed the problem by using the openmediavault interface (under Network->Interfaces->add)
> Hope this helps someone.

Thanks but these were virgin Raspbian installs.

enjay870, 2021-03-15:
I have same issue. How do you access the openmediavault interface when you do not have network?

StuartRoberts40, 2021-03-25:
> Hello, I had the same problem. Turns out some software packages, in my case OpenMediaVault, interfere with raspbian's networking services.
> I fixed the problem by using the openmediavault interface (under Network->Interfaces->add)
> Hope this helps someone.

SOLVED!
erikoui - Thank you so much! This was the fix for me. I had been going mad trying to find an answer for months.

Fam-of3, 2021-07-08:
I just tried setting up wifi using raspi-config, this is after a complete new completely fresh download of "RaspberryOS" on fresh brand new "Rpi Zero W" and am receiving the exact same issue, raspi-config is "unable to communicate with wpa_supplicant"... only access I have to my pi is via USB.ssh

This occurs with 'RaspberryOS -Recommended', 'RaspberryOS lite', and with 'RaspberryOS Full' failed to work at all on "Pi Zero W"

enjay870, 2021-07-08:
It seems that updated software packages can interfere.  I solved the problem using buster-lite 2019 OS without updates.  Now it works.  I have also blocked all OMV updates.  Again, not a satisfactory solution but it worked for me.

JiFengQAQ, 2021-10-09:
This works for me.

[[Can't set Wifi country (could not communicate to wpa_supplicant) - Q Light Controller+ (qlcplus.org)](https://www.qlcplus.org/forum/viewtopic.php?t=12468)]()

> ### [Re: Can't set Wifi country (could not communicate to wpa_supplicant)](https://www.qlcplus.org/forum/viewtopic.php?p=54476&sid=35ec1a2b3de5e0917abf556ad8bacafc#p54476)
>
> [Post](https://www.qlcplus.org/forum/viewtopic.php?p=54476&sid=35ec1a2b3de5e0917abf556ad8bacafc#p54476) by **[djcflo](https://www.qlcplus.org/forum/memberlist.php?mode=viewprofile&u=4630&sid=35ec1a2b3de5e0917abf556ad8bacafc)** » Mon Nov 26, 2018 7:01 pm
>
> I found success in doing it in two places (using a Rpi 3B+)
>
> over SSH, or on the unit via command lin type in
>
> ```
> sudo nano /etc/wpa_supplicant/wpa_supplicant-wlan0.conf
> ```
>
> hit enter to open text editor. it's a blank document. paste in
>
> ```
> ctrl_interface=/run/wpa_supplicant
> update_config=1
> country=US
> ```
>
> control X to quit, Y to save, enter to get back to command line
> sudo shutdown -h now
>
> 2nd place we have to do it:
> type in
>
> ```
> sudo nano /etc/wpa_supplicant/wpa_supplicant.conf
> ```
>
> hit enter
>
> in document, directly under ctrl_interface add
>
> ```
> country=US
> ```
>
> populate the fields with your home network info
>
> ```
> ssid=”home wifi name”
> psk=”home wifi password”
> ```
>
> control X to quit, Y to save, enter to get back to command line
>
> sudo reboot