raspibolt/raspibolt #457: Macroons not visible

hello9312, 2019-06-19:
Hello, 

Everything is running fine until i get to:
Check if permission files admin.macaroon and readonly.macaroon have been created.
_$2 sudo ls -la /home/bitcoin/.lnd/data/chain/bitcoin/testnet/_

This is my log:

admin@raspberrypi:~ $ sudo ls -la /home/bitcoin/.lnd/
total 28
drwxr-xr-x 4 bitcoin bitcoin 4096 Jun 19 21:04 .
drwxr-xr-x 5 bitcoin bitcoin 4096 Jun 19 18:10 ..
drwx------ 4 bitcoin bitcoin 4096 Jun 19 21:12 data
-rw-r--r-- 1 bitcoin bitcoin  439 Jun 19 21:04 lnd.conf
drwx------ 3 bitcoin bitcoin 4096 Jun 19 21:04 logs
-rw-r--r-- 1 bitcoin bitcoin  741 Jun 19 21:04 tls.cert
-rw------- 1 bitcoin bitcoin  227 Jun 19 21:04 tls.key
admin@raspberrypi:~ $ sudo ls -la /home/bitcoin/.lnd/data/chain/bitcoin/testnet/
total 24
drwx------ 2 bitcoin bitcoin  4096 Jun 19 21:12 .
drwx------ 3 bitcoin bitcoin  4096 Jun 19 21:12 ..
-rw------- 1 bitcoin bitcoin 16384 Jun 19 21:12 wallet.db
admin@raspberrypi:~ $ cd /home/bitcoin/
admin@raspberrypi:/home/bitcoin $ sudo cp --parents .lnd/data/chain/bitcoin/testnet/admin.macaroon 
cp: missing destination file operand after '.lnd/data/chain/bitcoin/testnet/admin.macaroon'
Try 'cp --help' for more information.

daniel-vera-g, 2019-06-19:
Did you create your wallet with:

`lncli --network=testnet create`

If so, after the official documentation you have to first start lnd:

> On startup, lnd checks to see if the admin.macaroon, readonly.macaroon and invoice.macaroon files exist. If they don't exist, lnd updates its database with a new macaroon ID, generates the three files admin.macaroon, readonly.macaroon and invoice.macaroon, all with the same ID.

hello9312, 2019-06-20:
All sorted, the nat=true line in lnd.conf was clashing with my router upnp settings. once deleted it worked.

closed 2019-06-20 by hello9312