raspibolt/raspibolt #458: Channel balance question.

CryptoAstroNut, 2019-06-27:
when I first started my raspibolt I only added a small amount of BTC to test out. (.008 btc) Everything was up and running and working fine with the auto settings. I had 3 channels open. I had an issue that forced me to close all my channels (forced them closed) I was able to get everything fixed and back up and running but now my balance is only (.004btc) When I look up all channels I have 3 active channels and 4 passive. When I look up my closed channels I see 1 cooperative close and 2 local_force_close. The forced closed have time lock balances that seem to add up to the missing BTC. Is there a way to retieve this balance back?

daniel-vera-g, 2019-06-27:
You've to wait till the timelock has passed. Also remember, that closing & opening channels also mean paying onchain transaction fees. They could be higher making a force close.

CryptoAstroNut, 2019-06-27:
Do you know how long the time lock is for or how to see when it will unlock? I closed the channels 2 months ago.

daniel-vera-g, 2019-06-27:
> Do you know how long the time lock is for or how to see when it will unlock? I closed the channels 2 months ago.

Hmm, you probably already have your funds back. Those have probably already been used to open your 8 new channels.  
2 months are too long, I don't think your funds are still in the timelock.

CryptoAstroNut, 2019-06-28:
I thought that too but when I check channel balance it returns 239379 and that is far off from what it should be since I have my setting to use 90% of wallet balance for channels.

Stadicus, 2020-02-28:
Closing as "too old" :-) Feel free to reopen if still relevant.

closed 2020-02-28 by Stadicus