raspibolt/raspibolt #460: UFW question

laddieri, 2019-06-28:
When adding exceptions to ufw, how exactly do I allow devices from within my local network? On some of the "BONUS" tutorials it instructs adding....

sudo ufw allow from 192.168.0.0/16 to any port 10009 comment 'allow LND grpc from local LAN'

But most of the devices on my LAN have an IP address like, 192.168.1.149

Should I change the ufw rule to allow from "192.168.1.0/16 ?  

What exactly does the "/16" or "/24" do?

Stadicus, 2019-07-04:
Yes, you need to adjust the first three parts of the ip address range (e.g. 192.168.1...) to your own network if you are using the /24 subnetmask. This means you are allowing all 256 ips from the fourth part (192.168.1.*) to connect.

I started using the /16 subnet mask, as this corresponds to 192.168.*.*
see https://www.kthx.at/subnetmask/

laddieri, 2019-07-04:
Thanks! So just to check if I understand correctly, the /16 subnet mask is less restrictive and allows any IP address from 192.168.*.* where /24 only allows from 192.168.0.* (or 192.168.1.* if I use 192.168.1. instead.)

Also, do you have any idea why I would continually receive "connection refused" errors when trying to access the raspibolt's bitcoin grpc via port 8332 from a desktop computer in the same LAN?

Stadicus, 2020-02-28:
Closing as "too old" :-) Feel free to reopen if still relevant.

closed 2020-02-28 by Stadicus