raspibolt/raspibolt #472: Regular Guide

Kartereo, 2019-07-28:
Would it be possible to create a regular guide for ubuntu? I am trying to follow the guide through here but keep getting stuck in places due to it being made for the pi. I think most people would be trying to follow a bitcoin node guide to install on a regular computer. I have found other guides but none are as thorough, use multiple users or included the extras. Your guide is the best out there! Just for reference I am trying to follow this guide to install on a ASRock J4105-ITX.

daniel-vera-g, 2019-07-28:
Well, I personally find this guide relatively easy to follow for others distributions & systems. This with the requirement, that you understand your system. 
The next thinkable idea, would be a debian based server documentation. Nevertheless, there is the issue that people will have different system requirements and OS.

Kineticmb, 2019-07-30:
The Raspi guide is like for like to turn an Odroid running Ubuntu into a LN node but with the exception of:

"Raspi Config" section cannot be used of course but you can use the section 5 from this old Stadicus guide instead:

https://medium.com/@stadicus/perfect-low-cost-%EF%B8%8Flightning%EF%B8%8F-node-4c2f42a4ff7b 

- Section 3 (For downloading the software)
- Section 5 MIXED with the raspi bolt guide as follows:

# change root password to [password A]
$ passwd 
# update the operating system
$ apt update
$ apt upgrade
$ apt dist-upgrade
$ apt install linux-image-xu3 
# answer [y], then [no] (do not abort)
# install some additional software
$ sudo apt-get install htop git curl bash-completion jq dphys-swapfile dirmngr --install-recommends
# set time zone & localization
$ dpkg-reconfigure tzdata
$ dpkg-reconfigure locales

# change hostname (replace "odroid" with "NAME OF YOUR CHOOSING"
$ nano /etc/hostname
$ nano /etc/hosts
# disable Swap file (this would degrade the MicroSD card)
$ swapoff --all

After getting to this point you continue with the RaspiBolt guide line for line other that when you need to enter hostname anywhere you put in the name you have chosen above.

Stadicus, 2019-12-19:
I am planning to create a "hardware decisioni tree" in the beginning, that would contain the changes necessary for other platforms / distributions. Odroid, Pine64 and regular Ubuntu seem like obvious choices.

If you have any specifc things you get stuck at, it would be helpful to list them here, so I can make sure these are addressed.

Kineticmb, 2019-12-22:
Fantastic idea!

Kartereo, 2020-05-09:
I will await the hardware decision tree. This is just too complicated for me after trying multiple times. Thank for you work!

closed 2020-05-09 by Kartereo