raspibolt/raspibolt #487: Tor configuration using `streamisolation`

bjarnemagnussen, 2019-09-16:
Hi,

I just wanted to mention that in the official LND documentation regarding Tor (https://github.com/lightningnetwork/lnd/blob/master/docs/configuring_tor.md) they advise to activate `streamisolation` like this:

```
[Tor]
tor.active=1
**tor.streamisolation=true**
tor.socks=localhost
tor.v3=true
```

Is there any specific reason for why inside the Raspibolt Tor documentation (`raspibolt_69_tor.md`) the option `streamisolation` is ignored?

Septem151, 2019-09-23:
According to the commit dates that `streamisolation` was added to the lnd.conf documentation - https://github.com/lightningnetwork/lnd/commit/4cba9dc8d3bae93f4e3478c46dca8b41cfb1b619#diff-2dad73b2f802ce6806e12e8d5dbeeaaf - this addition predates the creation of the RaspiBolt guide. 

@Stadicus should Tor users be setting `tor.streamisolation=true` in the lnd.conf file when running RaspiBolt, or is there an optimization issue with rPi's that prevents setting to true?

kristapsk, 2019-11-10:
Didn't notice this issue before, but I have addressed this in #496. Haven't noticed any performance issues with this with RPi 3 so far.

Stadicus, 2019-11-11:
Thanks, just merged it.

closed 2019-12-16 by bjarnemagnussen