raspibolt/raspibolt #488: Is it possible to implement this guiide with a running Raspiblitz?

lucbran, 2019-09-17:
Hi,
I start informing I am not skilled developer but folowing Raspiblitz guide, I setup a running Lightning network full node with Raspiblitz.
I understand I cannot use it as a normal Bitcoin full node (for example to validate my Ledger transactions) so I found this guide and reading it, I have some doubts which would be grateful if someone can clear:
1. can I implement Raspibolt instructions with my already operating Raspiblitz?
2. can I use my raspiblitz adding a new HDD with blockchain or alternatively, add an additional Raspberry PI to the already downloaded blockchain in my HDD?
3. none of the two are suitable: new raspberrypi+HDD should be setup as a different node.

Any chance to have a video tutorial on this one?
Thanks for clarifications

Septem151, 2019-09-23:
You can use your Raspiblitz as a Bitcoin Full Node, as it's a Bitcoin Full Node with Lightning. If it wasn't a Bitcoin Full Node, then you wouldn't be able to use Lightning.

Stadicus, 2019-12-19:
The RaspiBlitz is based on the version 1 of my guide, but automates most of it. It's a Lightning Node that also has a fully validating Bitcoin full node on it. 

What it lacks, and I find very important, is the feature you are describing, an easy way to use the Bitcoin client with hardware wallets.

The easiest way to do this is to also run an Electrum server on top of Bitcoin Core, and then use the Electrum wallet with your own node and a hardware wallet. That works with the Electrum Personal Server (works with a Pi3, see my bonus guide), or if you have a Raspberry Pi 4 just checkout the updated version 2 of this guide, as it now has a full Electrum server built in by default (which does not work on a Pi3, though).

closed 2020-02-28 by Stadicus