raspibolt/raspibolt #490: Raspberry PI 4 (4GB) and Blockstream Elements (isn't a issue but a new proposal)

Amiga500, 2019-09-24:
Why not add a guide for all the people with Raspberry Pi 4 (4GB) and new blockstream elements?

https://github.com/ElementsProject/elements/releases/download/elements-0.18.1/elements-0.18.1-arm-linux-gnueabihf.tar.gz

I have downloaded it but i have some issue in order to make it functional.

Amiga500, 2019-09-24:
bitcoin@raspberrypi:~ $ elementsd
2019-09-24T14:36:47Z Elements Core version elements-0.18.1 (release build)
2019-09-24T14:36:47Z Validating signatures for all blocks.
2019-09-24T14:36:47Z Setting nMinimumChainWork=0000000000000000000000000000000000000000000000000000000000000000
2019-09-24T14:36:47Z Using the 'standard' SHA256 implementation
2019-09-24T14:36:47Z Default data directory /home/bitcoin/.elements
2019-09-24T14:36:47Z Using data directory /home/bitcoin/.elements/liquidv1
2019-09-24T14:36:47Z Config file: /home/bitcoin/.elements/elements.conf (not found, skipping)
2019-09-24T14:36:47Z Using at most 125 automatic connections (1024 file descriptors available)
2019-09-24T14:36:47Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-09-24T14:36:47Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-09-24T14:36:47Z Using 32 MiB out of 32 requested for rangeproof cache, able to store 1048576 elements
2019-09-24T14:36:48Z Using 32 MiB out of 32 requested for surjectionproof cache, able to store 1048576 elements
2019-09-24T14:36:48Z Using 4 threads for script verification
2019-09-24T14:36:48Z scheduler thread start
2019-09-24T14:36:48Z HTTP: creating work queue of depth 16
2019-09-24T14:36:48Z No rpcpassword set - using random cookie authentication.
2019-09-24T14:36:48Z Generated RPC authentication cookie /home/bitcoin/.elements/liquidv1/.cookie
2019-09-24T14:36:48Z HTTP: starting 4 worker threads
2019-09-24T14:36:48Z Using wallet directory /home/bitcoin/.elements/liquidv1/wallets
2019-09-24T14:36:48Z init message: Verifying wallet(s)...
2019-09-24T14:36:48Z Using BerkeleyDB version Berkeley DB 4.8.30: (April  9, 2010)
2019-09-24T14:36:48Z Using wallet /home/bitcoin/.elements/liquidv1/wallets
2019-09-24T14:36:48Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.elements/liquidv1/wallets/database ErrorFile=/home/bitcoin/.elements/liquidv1/wallets/db.log
2019-09-24T14:36:48Z init message: Loading banlist...
2019-09-24T14:36:48Z Cache configuration:
2019-09-24T14:36:48Z * Using 2.0 MiB for block index database
2019-09-24T14:36:48Z * Using 8.0 MiB for chain state database
2019-09-24T14:36:48Z * Using 440.0 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2019-09-24T14:36:48Z init message: Loading block index...
2019-09-24T14:36:48Z Opening LevelDB in /home/bitcoin/.elements/liquidv1/blocks/index
2019-09-24T14:36:48Z Opened LevelDB successfully
2019-09-24T14:36:48Z Using obfuscation key for /home/bitcoin/.elements/liquidv1/blocks/index: 0000000000000000
2019-09-24T14:36:48Z LoadBlockIndexDB: last block file = 0
2019-09-24T14:36:48Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=1, size=732, heights=0...0, time=2011-02-02...2011-02-02)
2019-09-24T14:36:48Z Checking all blk files are present...
2019-09-24T14:36:48Z Opening LevelDB in /home/bitcoin/.elements/liquidv1/chainstate
2019-09-24T14:36:48Z Opened LevelDB successfully
2019-09-24T14:36:48Z Using obfuscation key for /home/bitcoin/.elements/liquidv1/chainstate: 8f9c364217bd3929
2019-09-24T14:36:48Z Loaded best chain: hashBestChain=1466275836220db2944ca059a3a10ef6fd2ea684b0688d2c379296888a206003 height=0 date=2011-02-02T23:16:42Z progress=0.000
2019-09-24T14:36:48Z init message: Rewinding blocks...
2019-09-24T14:36:48Z init message: Verifying blocks...
2019-09-24T14:36:48Z  block index              47ms
2019-09-24T14:36:48Z init message: Loading wallet...
2019-09-24T14:36:48Z BerkeleyEnvironment::Open: LogDir=/home/bitcoin/.elements/liquidv1/wallets/database ErrorFile=/home/bitcoin/.elements/liquidv1/wallets/db.log
2019-09-24T14:36:48Z [default wallet] nFileVersion = 180100
2019-09-24T14:36:48Z [default wallet] Keys: 2001 plaintext, 0 encrypted, 2001 w/ metadata, 2001 total. Unknown wallet records: 0
2019-09-24T14:36:48Z [default wallet] Wallet completed loading in             168ms
2019-09-24T14:36:48Z [default wallet] setKeyPool.size() = 2000
2019-09-24T14:36:48Z [default wallet] mapWallet.size() = 0
2019-09-24T14:36:48Z [default wallet] mapAddressBook.size() = 0
2019-09-24T14:36:48Z mapBlockIndex.size() = 1
2019-09-24T14:36:48Z nBestHeight = 0
2019-09-24T14:36:48Z torcontrol thread start
2019-09-24T14:36:48Z Bound to [::]:7042
2019-09-24T14:36:48Z Imported mempool transactions from disk: 0 succeeded, 0 failed, 0 expired, 0 already there
2019-09-24T14:36:48Z Bound to 0.0.0.0:7042
2019-09-24T14:36:48Z init message: Loading P2P addresses...
2019-09-24T14:36:48Z Loaded 7 addresses from peers.dat  1ms
2019-09-24T14:36:48Z init message: Starting network threads...
2019-09-24T14:36:48Z net thread start
2019-09-24T14:36:48Z dnsseed thread start
2019-09-24T14:36:48Z addcon thread start
2019-09-24T14:36:48Z opencon thread start
2019-09-24T14:36:48Z msghand thread start
2019-09-24T14:36:48Z init message: Awaiting mainchain RPC warmup
2019-09-24T14:36:48Z ERROR: Failure connecting to mainchain daemon RPC: Could not locate mainchain RPC credentials. No authentication cookie could be found, and no mainchainrpcpassword is set in the configuration file (/home/bitcoin/.elements/elements.conf)
2019-09-24T14:36:48Z Error: ERROR: elementsd is set to verify pegins but cannot get valid response from the mainchain daemon. Please check debug.log for more information.

[... cut ...]
2019-09-24T14:36:48Z Shutdown: In progress...
2019-09-24T14:36:48Z addcon thread exit
2019-09-24T14:36:48Z torcontrol thread exit
2019-09-24T14:36:48Z dnsseed thread exit
2019-09-24T14:36:48Z opencon thread exit
2019-09-24T14:36:48Z net thread exit
2019-09-24T14:36:48Z msghand thread exit
2019-09-24T14:36:48Z scheduler thread interrupt
2019-09-24T14:36:48Z Dumped mempool: 6e-06s to copy, 0.003887s to dump
2019-09-24T14:36:48Z [default wallet] Releasing wallet
2019-09-24T14:36:48Z Shutdown: done

Amiga500, 2019-09-24:
My actual bitcoin.conf is like this (and i have to do something similar to elements.conf too):

# RaspiBolt: bitcoind configuration
# /home/bitcoin/.bitcoin/bitcoin.conf

# remove the following line to enable Bitcoin mainnet
#testnet=1

# Bitcoind options
server=1
daemon=1

# Connection settings
rpcuser=raspibolt
rpcpassword=PASSWORD_[B]

onlynet=ipv4
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=80
maxuploadtarget=5000

Amiga500, 2019-09-24:
created elements.conf with these lines:

mainchainrpcuser=raspibolt
mainchainrpcpassword=PASSWORD_[B]
mainchainrpchost=127.0.0.1
mainchainrpcport=28332

Now it seems works :)

Stadicus, 2019-12-19:
Yes, installation is pretty straight forward, but to honest I don't see the usage scenario going from there. It's meant to be used with the included wallet, would you just run the `qt-elements` on your regular computer and connect it to your node instead of downloading the chain there? Or how would you interact with it?

Amiga500, 2019-12-20:
Hey Stadicus, my idea was to downloading the chain on rpi4 too.. but i stopped.. it would be very nice to have Liquid on raspibolt #2 ;) i don't know of any full node that had integrated it yet.. probably the first one would be myNode or Raspibolt.. or Raspiblitz? :)

Stadicus, 2020-02-28:
Closing as out-of-scope for now, as I keep RaspiBolt accessible as a Beginner's guide. Happy to link to a tutorial from the Bonus section.

closed 2020-02-28 by Stadicus