raspibolt/raspibolt #495: LevelDB read failure: Corruption: not an sstable

ZebulonFrantzich, 2019-10-28:
I am running Bitcoin Core version v0.18.1 (release build), and I am running into the following error:

```
2019-10-28T14:45:27Z init message: Rewinding blocks...
2019-10-28T14:45:29Z init message: Verifying blocks...
2019-10-28T14:45:29Z Verifying last 6 blocks at level 3
2019-10-28T14:45:29Z [0%]...LevelDB read failure: Corruption: not an sstable (bad magic number)
2019-10-28T14:45:30Z Fatal LevelDB error: Corruption: not an sstable (bad magic number)
2019-10-28T14:45:30Z You can use -debug=leveldb to get more complete diagnostic messages
2019-10-28T14:45:30Z Fatal LevelDB error: Corruption: not an sstable (bad magic number)
2019-10-28T14:45:30Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2019-10-28T14:45:30Z Aborted block database rebuild. Exiting.
2019-10-28T14:45:30Z Shutdown: In progress...
2019-10-28T14:45:30Z scheduler thread interrupt
2019-10-28T14:45:30Z Shutdown: done
```

Has anyone else run into this issue? I have been running a node for the past year or so, and haven't had any issues with verifying blocks.

53645714n, 2019-10-28:
This is interesting. I had ext4 fs errors yesterday and after a restart the same error. Does your syslog say anything?

What I am doing: 

    $ sudo systemctl disable bitcoind.service
    $ bitcoin-cli stop
    $ sudo su bitcoin
    $ bitcoind -reindex
And wait....

ZebulonFrantzich, 2019-10-28:
@53645714n Thank you for your response! I do not see much useful info in the syslog:

```
Oct 28 10:18:17 systemd[1]: Starting Bitcoin daemon...
Oct 28 10:18:48 systemd[1]: Started Bitcoin daemon.
Oct 28 10:19:19 systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 10:19:19 systemd[1]: bitcoind.service: Unit entered failed state.
Oct 28 10:19:19 systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Oct 28 10:19:49 systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
Oct 28 10:19:49 systemd[1]: Stopped Bitcoin daemon.
```

I might have to reindex...

53645714n, 2019-10-28:
I had the exact same log. re-indexing could take a day or two, but then I can let you know if it worked.

53645714n, 2019-11-05:
@ZebulonFrantzich, did you manage to fix this? My node is still reindexing, I interfered a few times and that didn't help so now I am trying the patient approach..

ZebulonFrantzich, 2019-11-05:
@53645714n My node is reindexing as well. I am also trying the patient approach. It will probably take another week or so. I will let you know when it finishes.

ZebulonFrantzich, 2019-12-16:
@53645714n My node finished reindexing over the weekend, and everything seems to be running smoothly. Any luck on your end?

closed 2019-12-17 by ZebulonFrantzich

53645714n, 2019-12-17:
> @53645714n My node finished reindexing over the weekend, and everything seems to be running smoothly. Any luck on your end?

Same here, after reindexing everything was fine again. A day later, same error occured so now I'm buidling a new raspibolt hdd only.

openoms, 2019-12-17:
You might be getting problems with the SSD due to an adapter problem. See the explanation: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

Pick one of these adapters: https://github.com/rootzoll/raspiblitz/issues/691#issuecomment-520526004 or any which has been confirmed working with RPI4 + SSD