raspibolt/raspibolt #499: UPnP error when starting LND

Stealth0113, 2019-11-24:
I'm setting up an Bitcoin/LND node for the first time, the tutorial works very good. But when i'm starting LND i get an UPnP error. I saw several treats about this UPnP problem but non of them worked for me.

Here is the log when i started LND
[image]

Port Forwarding:
[image]
UPnP
[image]
And the UFW settings:
[image]

I really don't know what i'm doing wrong. I read the troubleshoot section and it works like a charm until i start LND.

Any suggestions where to look?

Stealth0113, 2019-11-24:
I've Bitcoin Core 0.19 installed today on my windows laptop. Set port 8333 to my laptop instead of the raspberry pi and now my ip is visible on Bitnodes.  So the problem isn't in my router but in the PI settings.

bruteforcecat, 2019-12-16:
I was also having the same problem but then it finds out my router is not using standard port 1900 for SSDP. you can check  if your firewall actually block those attempt from your router by `sudo dmesg | grep '\[UFW'`

You can try ufw disable and restart your lnd to see if it's working. If yes, you can set sudo ufw allow from 192.168.1.1(your router address) before finding out which port your router is using for upnp first.

Stadicus, 2019-12-19:
For LND to be reachable from the outside, I'd recommend just using Tor, as UPnP may not work in every network and can be a pain to set up correctly.

closed 2020-02-28 by Stadicus