raspibolt/raspibolt #502: incorporate a BTCPayServer installation

phenomeus, 2019-11-26:
Hi @Stadicus 

I'm running bitcoin/lightning over tor and am pretty happy with it. but I wish there would be a guide to install a btcpay server on top of it and using all your previous setup. 

I tried this guide but couldn't get it to use the existing bitcoin and lightning setup. 
https://freedomnode.com/blog/114/how-to-setup-btc-and-lightning-payment-gateway-with-btcpayserver-on-linux-manual-install

talking to Mario Dan and the btcpayserver people also brought no light on the issues. 

do you have the time (mostly not) and muse to tackle that? I would love to help and beta test if you wish

hope to hear from you soon

openoms, 2019-11-26:
Could look into this guide which works on the RaspiBlitz (with similar infrastructure to the RaspiBolt)
https://github.com/openoms/bitcoin-tutorials/blob/master/BTCPayServer/BTCPayServer_on_the_RaspiBlitz.md

The scripted version is more up to date and separated BTCPaySetver to the `btcpay` user: https://github.com/openoms/raspiblitz-extras/blob/master/config.scripts/bonus.btcpayserver.sh

Would be great to test and adapt either to the RaspiBolt too.

phenomeus, 2019-11-26:
thanks, I will have a look and try to get that working. never thought about raspiblitz and a guide there.

phenomeus, 2019-11-27:
@openoms I followed that guide and stuck at the end with this… 

502 Bad Gateway
nginx

any ideas? thank you

openoms, 2019-11-27:
Yeah, Nginx is a painpoint.
try:
 `sudo nginx -t`
it might give you some clues.
Make sure that your ports 80 and 443 are open and you are visiting the domain / dyndns you have set up.

phenomeus, 2019-11-27:
80 and 443 as tcp are open, dyndns is working. 

"nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful"

phenomeus, 2019-11-27:
also nbxplorer is around 25% memory and that's killing my pi 3b+ unresponsive… 

its already running pretty much with btc node, lnd, homebridge, mqtt, openvpn & tor, so I think I need to get it running and think about pi 4 with 2GB+ ram… :D

phenomeus, 2019-11-27:
tomorrow I will try to recreate the installation step by step from your install script for RaspiBlitz. now my ambition is awake…

phenomeus, 2019-11-28:
@openoms I did it, I reinstalled everything step by step from the script. but I got the same error… and then: I saw the comment at the beginning of the script to open the port 80,443 and 9735. 

and ** 9735** wasn't open. BAM! its working, got the BTCPay Frontend visible. now lets try to get everything hooked up together.

openoms, 2019-11-28:
@phenomeus interesting. 9735 should only affect the availability of the Lightning node. 80 has HTTP, 443 HTTPS which is served with Nginx.

phenomeus, 2019-11-28:
@openoms, so nbxplorer is still syncing… and eating bunch of memory on my pi 3b+. I hope it will end as soon its synced and I can fire up lnd and BTCPayServer again. 

quick question: any idea how to switch nginx with BTCPayServer on port 8000? I need lighthttpd service on 80 for pihole

csh2000, 2020-01-26:
Hello, also interested in running BTCPay Server on top of Raspibolt... @phenomeus , did you finally got it?.
Any good guide for this?
Thanks and congrats @Stadicus for this guide,  v.2 is awsome!!

phenomeus, 2020-01-29:
I stopped it on my pi3 because its just underpowered. got myself a pi4 4gb and also doing a migration over to raspiblitz which includes btcpayserver

closed 2020-01-29 by phenomeus