raspibolt/raspibolt #503: JoinMarket on RaspiBolt

kristapsk, 2019-12-02:
I am working on tutorial how to get JoinMarket working on RaspiBolt. Still consider it as work in progress, but is useful already, I think. https://github.com/kristapsk/raspibolt-extras/blob/master/joinmarket.md

Opening issue for two reasons: 1) for more RaspiBolt users to know about it, 2) should I send it in future PR as a part of RaspiBolt guide or just as a link in "Bonus Section > Even more Extras"?

Stadicus, 2019-12-03:
Awesome! Currently swamped with other things, but I'll have a look as soon as I can.

raindogdance, 2019-12-13:
FWIW, I followed that guide and can confirm that it worked for me just fine :) Thanks @kristapsk!  

Small suggestion: 

In [this](https://github.com/kristapsk/raspibolt-extras/blob/master/joinmarket.md#checking-wallet-balance-and-history) section I would also add `python wallet-tool.py wallet.jmdat display` command, because it allows the user to see the balances in different mixdepths at one glance.

Stadicus, 2020-02-28:
For now, I'll try to keep the RaspiBolt guide to the essentials (Bitcoin, Electrum, Lightning) and have it accessible as a "Beginner's guide". So not planning to add JoinMarket, which is complex to setup and operate, as a main part. Happy to link to your guide from the bonus section, though.

Are you planning to keep it updated @kristapsk?

kristapsk, 2020-02-28:
Yes, I'm planning to update the guide, currently waiting for next JoinMarket release which separates code and data (wallets, logs) directories, which makes setup and upgrade more easy (code on SD card, logs/wallets on HDD).

Stadicus, 2020-02-28:
Cool! Will link it from the Bonus page, ok?

kristapsk, 2020-02-28:
Probably I should finish it and then send a PR when I feel it's ready. Or you want to link WIP? Also, it's currently for RaspiBolt 1, need to check what's probably changed for RaspiBolt 2 (and probably need to order RPi 4 for myself, to test it).

Stadicus, 2020-02-28:
Once you have everything ready and open the pull-request, I'll be happy to test it for the RaspiBolt2. There shouldn't be too many differences. One is that the external drive is now mounted to `/mnt/ext` instead of `/mnt/hdd`.

PR #581 (2020-03-26): Add link to JoinMarket guide for RaspiBolt
Tested this with RaspiBolt v1 on RPi 3, changed `/mnt/hdd` to `/mnt/ext` afterwards. Let me know if something else needs to be changed!

Closes #503.
closed 2020-03-26 by Stadicus