raspibolt/raspibolt #510: Inconsistency of path to `bitcoind` and `lnd`?

bjarnemagnussen, 2019-12-16:
I just saw that in the updates RaspiBolt (for RPi4), the installation path for `bitcoind` (https://github.com/Stadicus/RaspiBolt/blob/master/raspibolt_30_bitcoin.md#installation) has changed from `/usr/bin/local/` to `/usr/bin/`, while the installation path for `lnd` (https://github.com/Stadicus/RaspiBolt/blob/master/raspibolt_40_lnd.md#installation) stays the same at `/usr/bin/local`.

I was just wondering if that choice of different installation paths is on purpose?

Stadicus, 2019-12-16:
I changed it for `bitcoind` to stay as close to the reference documentation as possible, see https://github.com/bitcoin/bitcoin/blob/master/contrib/init/bitcoind.service

Did not change it for LND, might be something to consider.

Stadicus, 2019-12-16:
After some research and reading this [StackExchange explanation](https://unix.stackexchange.com/a/8658), I think all manually installed binaries belong into `/usr/local/bin`.

Merged https://github.com/Stadicus/RaspiBolt/pull/512/commits/11da1eda928f5d5ff90308c71fa23326f282e001 to fix that.

closed 2019-12-19 by Stadicus