raspibolt/raspibolt #513: LevelDB read failure:

warfield2016, 2019-12-18:
3.9MiB(215736txo)
2019-12-18T02:45:38Z UpdateTip: new best=0000000000000000000308a4793e144acd5d844caddcd3d5f4237600a930b12e height=587512 version=0x20000000 log2_work=90.898717 tx=439449855 date='2019-07-28T22:36:00Z' progress=0.904309 cache=24.0MiB(216247txo)
2019-12-18T02:45:38Z New outbound peer connected: version: 70015, blocks=607944, peer=4 (full-relay)
2019-12-18T02:45:38Z New outbound peer connected: version: 70015, blocks=608593, peer=5 (full-relay)
2019-12-18T02:45:39Z UpdateTip: new best=00000000000000000004394d237c2df67ba539501468af53033bc21ccb6b6144 height=587513 version=0x20400000 log2_work=90.898741 tx=439449948 date='2019-07-28T22:36:16Z' progress=0.904309 cache=24.1MiB(217265txo)
2019-12-18T02:45:39Z New outbound peer connected: version: 70015, blocks=608593, peer=6 (full-relay)
2019-12-18T02:45:41Z LevelDB read failure: IO error: /home/bitcoin/.bitcoin/chainstate/977013.ldb: Input/output error
2019-12-18T02:45:41Z Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/chainstate/977013.ldb: Input/output error
2019-12-18T02:45:41Z You can use -debug=leveldb to get more complete diagnostic messages
2019-12-18T02:45:41Z Error: Error reading from database, shutting down.
2019-12-18T02:45:41Z Error reading from database: Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/chainstate/977013.ldb: Input/output error

warfield2016, 2019-12-18:
Any ideas on how to solve the issue above?

Stadicus, 2019-12-19:
Seems like the Bitcoin blockchain database is corrupted. To get more info, you can add `debug=leveldb` to the `bitcoin.conf` file and start `bitcoind` again.

Not sure if this is recoverable. Easiest would probably be to reindex (if you're on a Pi4), or copy the blockchain from your regular computer again (on a Pi3).

closed 2020-02-28 by Stadicus